geerlingguy/pi-webcam #1: v4l2 camera caps

nitzmahone, 2020-09-24:
Here's the ask, as requested. ;) When you've got this connected to a Linux host, can you paste the output from `v4l2-ctl --all -L --list-formats-ext -d /dev/(whatever)`? I'm curious to see what all capabilities they've implemented... Thanks!

geerlingguy, 2020-09-30:
```
$ v4l2-ctl --all -L --list-formats-ext -d /dev/video0
Driver Info:
	Driver name      : uvcvideo
	Card type        : PI4 USB Device: UVC Camera
	Bus info         : usb-0000:01:00.0-1.3
	Driver version   : 5.4.51
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Media Driver Info:
	Driver name      : uvcvideo
	Model            : PI4 USB Device: UVC Camera
	Serial           : 100000000d2386db
	Bus info         : usb-0000:01:00.0-1.3
	Media version    : 5.4.51
	Hardware revision: 0x00000100 (256)
	Driver version   : 5.4.51
Interface Info:
	ID               : 0x03000002
	Type             : V4L Video
Entity Info:
	ID               : 0x00000001 (1)
	Name             : PI4 USB Device: UVC Camera
	Function         : V4L2 I/O
	Flags         : default
	Pad 0x01000007   : 0: Sink
	  Link 0x0200000d: from remote pad 0x100000a of entity 'Processing 2': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
	Width/Height      : 1920/1080
	Pixel Format      : 'MJPG' (Motion-JPEG)
	Field             : None
	Bytes per Line    : 0
	Size Image        : 460800
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 1920, Height 1080
	Default     : Left 0, Top 0, Width 1920, Height 1080
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Selection: crop_bounds, Left 0, Top 0, Width 1920, Height 1080, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 0.200 (1/5)
	Read buffers     : 0
                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=55 value=0
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=0 value=0
				0: Auto Mode
                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=55 value=0
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=0 value=0
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'MJPG' (Motion-JPEG, compressed)
		Size: Discrete 1920x1080
			Interval: Discrete 0.500s (2.000 fps)
```

I believe the resolution, framerate, etc. is controllable via the scripts included with the forked `uvc-gadget` repo. I'm going to look into making that simpler to modify next :)

geerlingguy, 2020-09-30:
See #4 for a continuation on that last thread.

closed 2020-09-30 by geerlingguy