geerlingguy/pi-webcam #4: Allow configuration of resolution, framerate, and brightness

geerlingguy, 2020-09-30:
At a minimum, I'd like to be able to set:

  - Resolution: Maybe default to 1280x720 (720p) so the Pi Zero doesn't drop frames as much?
  - Framerate: Default to 30 fps
  - Brightness: Default to 55

geerlingguy, 2020-09-30:
Looking at `multi-gadget.sh`, though, it is _not_ something made to be very malleable :-/

geerlingguy, 2020-09-30:
The simplest way to switch between 1080p and 720p at this point is to manually hack in this block in place of the 1080p text in multi-gadget.sh:

```bash
mkdir -p /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/720p
cat <<EOF > /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/720p/dwFrameInterval
5000000
EOF
cat <<EOF > /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/720p/wWidth
1280
EOF
cat <<EOF > /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/720p/wHeight
720
EOF
cat <<EOF > /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/720p/dwMinBitRate
29491200
EOF
cat <<EOF > /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/720p/dwMaxBitRate
29491200
EOF
cat <<EOF > /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/720p/dwMaxVideoFrameBufferSize
1843200
EOF

# Remove the existing 1080p dir.
rm -f /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/1080p
```

geerlingguy, 2020-09-30:
I'm gonna look into creating a fork. Might try to reincorporate changes back but for now... there are just too many little things that are annoying me.

geerlingguy, 2020-09-30:
So I did fork the project more strongly, and I made it default to 720p, 30fps, 55 brightness for now (much smoother on the Pi Zero, no dropped frames, yay!).

See my downstream fork here: https://github.com/geerlingguy/uvc-gadget

The process, right now, to change resolution:

  1. Re-flash the Pi OS
  1. Run this playbook, but don't reboot at the end.
  1. Modify the uvc-gadget repo files on the Pi (log in and cd to `/home/pi/uvc-gadget`):
    - Change the piwebcam script, uncommenting the 1080p section and commenting the 720p section.
    - Change the multi-gadget.sh script, doing the same.
  1. Shut down / reboot / start using the Pi. On boot it should pick up the proper resolution in the piwebcam startup.

If you want to switch after that, since you can't delete the m/[resolution] directory even as `root`, you have to re-flash. There might be a workaround for that but I haven't figured it out yet...

dgsiegel, 2020-11-04:
All of these can be now configured with https://github.com/peterbay/uvc-gadget

geerlingguy, 2020-11-10:
@dgsiegel - good find! It doesn't seem like the changes are all documented there; do you know if @peterbay is working on documenting things somewhere already? Honestly, for this project, I'd be happy to make it a wrapper to be able to use any of the (dozens?) of uvc-gadget forks.

The only reason I made my fork was to try to get things a little bit more stable for the Pi Zero. It would be great if this wide community of Pi Webcam people could focus our efforts on polishing up one of these forks :)

peterbay, 2020-11-10:
@geerlingguy - documentation and wiki pages are been on my TODO list. Both of these tasks are limited by my free time and my knowledge of English.

peterbay, 2020-11-11:
I started writing documentation. This is the beginning and I will gradually add more information - [uvc-gadget - documentation](https://github.com/peterbay/uvc-gadget/tree/master/doc)

geerlingguy, 2020-11-11:
@peterbay - That is awesome, thanks so much for moving things forward in your repo!

dgsiegel, 2020-11-11:
Also check out https://github.com/showmewebcam/showmewebcam where we try to bundle the efforts

geerlingguy, 2020-12-01:
From a blog post comment from David Hunt:

> When I’m logged in over serial, I can issue the following commands to adjust various things…
> ```
> /usr/bin/v4l2-ctl -c auto_exposure=0
> /usr/bin/v4l2-ctl -c auto_exposure_bias=8
> /usr/bin/v4l2-ctl -c contrast=20
> /usr/bin/v4l2-ctl -c video_bitrate=25000000
> ```

dgsiegel, 2020-12-01:
We are using a new tool for this: https://github.com/peterbay/camera-control

Will be in the next showmewebcam release

geerlingguy, 2020-12-01:
Very nice. I'm getting the feeling I'll have to do a follow-up video on showmewebcam...

dgsiegel, 2020-12-01:
Haha wait for the next release then :)

HunterAP23, 2020-12-07:
Just wanted to comment saying that I've tested this out on my Pi 4B (been trying to figure out a solution myself, can't thank you enough for figuring a lot of the issues out).

Installing Raspbian 2020-02-07, not updating the system, and setting up this repo went perfectly well, and 720p video actually works perfectly. The issue comes when I change the `piwebcam` file from the `uvc-gadget` repo to use 1080p video instead of 720p, and here's where the issues start.

1. The camera will appear on my Windows machine as a UVC device, but not actually have any output when viewed. Some programs like OBS just see a black screen, but others like the Windows 10 Camera app just error out.
2. Can't restart the service when changing the `piwebcam` file - you have to specifically restart the entire Pi, which is something I believe you said you were working on. This is an issue present with all other `uvc-gadget` repos I've tried on the Pi 4B
3. Not actually an issue, but interesting thing to point out is that when I use the camera on a Pi Zero W, stopping or restarting the piwebcam service causes the entire Pi Zero W to crash, and it only works again after unplugging it from power. This doesn't seem to happen on the Pi 4B, but it might be due to me now running the 2020-02-07 build.

kt-personal, 2020-12-10:
Thanks for all the work on this, looking forward to updates. I got this working on my HQ+pi4B as well in 720p with 2020-02-07 build, and look forward to 1080p when that's sorted. I currently have an issue where the camera is plugged in to my Win 10 based Lenovo laptop, After a while, when the camera is not in use, it seems to go to sleep and won't wake when called by an app like Teams or Camera. I don't have any sleep settings turned on for the laptop, Does the pi have any by default?

geerlingguy, 2020-12-10:
@kt-personal - The Pi doesn't have any kind of sleep functionality built in, but maybe it could be locking up or something... you'd have to log into it and see by monitoring its logs (via SSH over WiFi or serial connection).

Note that if the laptop drops power to its USB ports at some point (e.g. doing its own sleep), the Pi would shut off, then need to boot for ~30s again.

if-p-then-q, 2021-01-04:
@geerlingguy - This is a cool project!  720p on a pi4B running Raspbian 2020-02-07 works great on both my Windows 10 and Linux devices.  I've been using this setup for about a month and it's been very stable.

I'm also here to comment that I'm having the same issue as @HunterAP23 when I try to stream at 1080p on the pi4B.  I re-flashed the 2020-02-07 build, ran the playbook, and modified the relevant scripts.  Upon reboot, Windows recognizes the pi as a UVC Camera in device manager, but none of my apps display a video feed from the pi.

1.  Teams recognizes the pi as a UVC Camera, but the output is just a black screen.
2.  Zoom recognizes the pi as a UVC Camera, but switching to this camera option locks up Zoom for several minutes and eventually results in a blank grey screen.
3.  The Windows Camera App throws the following error message after launching: 0xA00F4271<MediaCaptureFailedEvent> (0x800703E3).  No video output appears on the screen.

NazarovVassil, 2021-01-25:
Jeff,

I just tested the latest [showmewebcam](https://github.com/showmewebcam/showmewebcam) on a Pi Zero W. It is fantastic:
- All resolutions are available for selection from within Windows up to 1080p/30.
- Boots for under 10 s.
- You can access the terminal on COM4 with root/root
- In the terminal, execute /usr/bin/camera-ctl to access all camera settings.

You should definitely do another video.

Thanks, all!

kravens, 2021-02-17:
Cool project! Just built my own using your guide.
Being able to the brightness in a simple command or config file would help me a lot, as the raspistill command has no effect on the webcam settings.

acidthefurry, 2022-01-14:
@AudiThisWorld it works on the pi 2 if you install it on the lite image with SSH

AudiThisWorld, 2022-01-14:
> @AudiThisWorld it works on the pi zero 2 if you install it on the lite image with SSH

Thanks for that. 

I have tried raspios-buster-armhf-lite version, and my M1 Max MacBook Pro does not see the camera. My Windows 11 PC does not see it either. 

Unfortunately, I have spent hours and hours on this, and I just might have to go back to my Logitech camera with digital zoom, which is a shame because I have this high-quality Raspberry Pi camera looking great with all of my 3D printed parts. =(

dgsiegel, 2022-01-14:
> I have tried raspios-buster-armhf-lite version, and my M1 Max MacBook Pro does not see the camera. My Windows 11 PC does not see it either.

It's most likely this bug which we fixed over at the showmewebcam project: https://github.com/showmewebcam/showmewebcam/pull/169

> Unfortunately, I have spent hours and hours on this, and I just might have to go back to my Logitech camera with digital zoom, which is a shame because I have this high-quality Raspberry Pi camera looking great with all of my 3D printed parts. =(

There's some progress in making this work on the pi zero 2 here: https://github.com/showmewebcam/showmewebcam/pull/173

geerlingguy, 2022-01-14:
Indeed, I think it might be best for people to consider using showmewebcam (as it has a larger community and more development going on), and I still have it on my schedule to _someday_ do a video on it in particular. I just have too many other things too :P

fullmakeralchemist, 2022-07-20:
Hi I'm using a windows with the Raspberry Pi Zero W, but it looks that is not working I tried 4 times and eachone I try to follow the issue, or comments that I see but I'm not sure why is still not working some help please?