geerlingguy/pi-webcam #5: Not working on Raspberry Pi 4 model B currently

geerlingguy, 2020-09-30:
I wanted to be able to have this project work great on the Pi 4, even though it makes the footprint a little larger... but it seems to not work so well. When you start the service, the system just halts, no input, no log output, nothing. And when you have it enabled on boot, the system gets to the step `polkit.service` (according to the GUI boot screen) and then does the same (just... stops).

The green LED goes on and remains steady, lit until you physically power off the Pi. Nothing responds to input.

Strange thing, and it looks like I'm not the only one who's experienced this—see https://gist.github.com/justinschuldt/36469e2a89d95ef158a8c4df091e9cb4#gistcomment-3467068

geerlingguy, 2020-10-01:
Trying on the Pi 3B, it got stuck at ssh.service on the screen, but the ACT LED never stayed on green.

geerlingguy, 2020-10-01:
2B is giving different behaviors on different boots. It gets past the 'Welcome to the Desktop' splash screen sometimes, and it's ACT LED doesn't stay lit green either.

geerlingguy, 2020-10-01:
Just for fun I tried the original Pi 1 and it booted all the way, but I don't believe its USB power port can double as OTG. I'm testing with an original A+ model now...

geerlingguy, 2020-10-01:
On my Pi 4, after starting off the `piwebcam` service:

```
[ 1294.311630] configfs-gadget gadget: uvc: uvc_function_bind()
[ 1294.311798] dwc2 fe980000.usb: bound driver configfs-gadget
[ 1315.360689] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[ 1315.360699] rcu:   3-...0: (1 GPs behind) idle=53e/1/0x40000000 softirq=6427/6430 fqs=822 
[ 1315.360704]  (detected by 0, t=2102 jiffies, g=11585, q=1055)
[ 1315.360709] Sending NMI from CPU 0 to CPUs 3:
[ 1325.361499] rcu: rcu_sched kthread starved for 1001 jiffies! g11585 f0x0 RCU_GP_WAIT_FQS(5) ->state=0x402 ->cpu=0
[ 1325.361502] rcu: RCU grace-period kthread stack dump:
[ 1325.361506] rcu_sched       I    0    10      2 0x00000000
[ 1325.361511] Backtrace: 
[ 1325.361525] [<c0a54c00>] (__schedule) from [<c0a5545c>] (schedule+0x5c/0xcc)
[ 1325.361531]  r10:eff0d440 r9:00018c9d r8:00018c9e r7:c1203d00 r6:c1204f88 r5:ef8f6000
[ 1325.361535]  r4:ef8f8f40
[ 1325.361542] [<c0a55400>] (schedule) from [<c0a59988>] (schedule_timeout+0x250/0x31c)
[ 1325.361546]  r5:ef8f7ef4 r4:eff0d440
[ 1325.361555] [<c0a59738>] (schedule_timeout) from [<c02953dc>] (rcu_gp_kthread+0x5ec/0xb04)
[ 1325.361561]  r10:00000000 r9:c1203d00 r8:c1205050 r7:c1210f58 r6:00000001 r5:c1210dc0
[ 1325.361564]  r4:00000000
[ 1325.361571] [<c0294df0>] (rcu_gp_kthread) from [<c0244e90>] (kthread+0x13c/0x168)
[ 1325.361574]  r7:00000000
[ 1325.361580] [<c0244d54>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[ 1325.361584] Exception stack(0xef8f7fb0 to 0xef8f7ff8)
[ 1325.361589] 7fa0:                                     00000000 00000000 00000000 00000000
[ 1325.361593] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 1325.361598] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 1325.361602]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0244d54
[ 1325.361606]  r4:ef86ecc0 r3:ffffffff
[ 1378.411728] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[ 1378.411738] rcu:   3-...0: (1 GPs behind) idle=53e/1/0x40000000 softirq=6427/6430 fqs=3387 
[ 1378.411743]  (detected by 0, t=8407 jiffies, g=11585, q=2146)
[ 1378.411748] Sending NMI from CPU 0 to CPUs 3:
[ 1441.462725] rcu: INFO: rcu_sched detected stalls on CPUs/tasks:
[ 1441.462735] rcu:   3-...0: (1 GPs behind) idle=53e/1/0x40000000 softirq=6427/6430 fqs=6041 
[ 1441.462740]  (detected by 1, t=14712 jiffies, g=11585, q=2168)
[ 1441.462745] Sending NMI from CPU 1 to CPUs 3:
...
```

Screenshotted:

[image]

And over in dmesg seeing a lot of:

```
Oct 01 17:55:37 raspberrypi systemd[1]: Started Session c3 of user pi.
Oct 01 17:55:39 raspberrypi systemd[1]: getty@ttyGS0.service: Succeeded.
Oct 01 17:55:39 raspberrypi systemd[1]: getty@ttyGS0.service: Service has no hold-off time (RestartSec=0), scheduling re
Oct 01 17:55:39 raspberrypi systemd[1]: getty@ttyGS0.service: Scheduled restart job, restart counter is at 1.
Oct 01 17:55:39 raspberrypi systemd[1]: Stopped Getty on ttyGS0.
Oct 01 17:55:39 raspberrypi systemd[1]: Started Getty on ttyGS0.
Oct 01 17:55:39 raspberrypi agetty[1025]: /dev/ttyGS0: cannot open as standard input: No such file or directory
Oct 01 17:55:42 raspberrypi dhcpcd[434]: wlan0: no IPv6 Routers available
Oct 01 17:55:43 raspberrypi dhcpcd[434]: eth0: no IPv6 Routers available
Oct 01 17:55:49 raspberrypi systemd[1]: getty@ttyGS0.service: Succeeded.
Oct 01 17:55:49 raspberrypi systemd[1]: getty@ttyGS0.service: Service has no hold-off time (RestartSec=0), scheduling re
Oct 01 17:55:49 raspberrypi systemd[1]: getty@ttyGS0.service: Scheduled restart job, restart counter is at 2.
Oct 01 17:55:49 raspberrypi systemd[1]: Stopped Getty on ttyGS0.
Oct 01 17:55:49 raspberrypi systemd[1]: Started Getty on ttyGS0.
Oct 01 17:55:49 raspberrypi agetty[1026]: /dev/ttyGS0: cannot open as standard input: No such file or directory
```

jensbocklage, 2020-10-08:
Had the same issues.
Then I found this: https://gist.github.com/justinschuldt/36469e2a89d95ef158a8c4df091e9cb4
I used this image instead of the latest one: Raspbian Buster Lite ### **2020-02-13**

Now it works.

blaueente, 2020-10-19:
Had the same issues, with a raspberry Pi 4, I also had this RCU error.
Similar to @jensbocklage downgrading helped.
However, being stuck at a lower kernel/operating system version is not a good option.

lennart, 2020-10-21:
just a note for those struggling to get the pi 4 (I think it is model B) to work following the [Directions for setting up a RaspberryPi to act as a generic USB webcam ](https://gist.github.com/justinschuldt/36469e2a89d95ef158a8c4df091e9cb4) even with the _older_ Raspbian Buster Lite 2020-02-13:

Do __not__ run `apt-get upgrade` as it will upgrade the kernel to 5.x which makes it hang again. Simply update and install git only. I had to reflash the sd card multiple times to get this right...

Right now using both the internal webcam of my laptop and the pi 4 with attached hq cam as a uvc device within obs studio on arch linux. I cannot yet confirm hangs, or stability issues as @blaueente describes, although when trying to stop uvc-gadget on the pi, it will fail and eventually try to remount the whole sdcard filesystem read-only and also become unresponsive (i.e. I have to power cycle it)

The pi is connected to network via wifi, I have not yet tested wether it can also connect via ethernet while being used as an uvc device.

Also note that, until I successfully started `uvc-gadget` I only got errors like (dmesg output on host):

```
[128791.859717] usb 1-6.2.1: new high-speed USB device number 39 using xhci_hcd
[128797.053057] usb 1-6.2.1: device descriptor read/64, error -110
[128812.626629] usb 1-6.2.1: device descriptor read/64, error -110
[...]
[128844.743085] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[128844.949699] usb 1-6.2.1: device not accepting address 41, error -62
[...]
[128850.076429] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[128855.409763] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[128855.616367] usb 1-6.2.1: device not accepting address 42, error -62
[128855.616571] usb 1-6.2-port1: unable to enumerate USB device
```
and only after starting uvc-gadget I got:

```
[128902.299754] usb 1-6.2.1: new high-speed USB device number 43 using xhci_hcd
[128902.413785] usb 1-6.2.1: New USB device found, idVendor=1d6b, idProduct=0104, bcdDevice= 1.00
[128902.413792] usb 1-6.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[128902.413796] usb 1-6.2.1: Product: PI4 USB Device
[128902.413798] usb 1-6.2.1: Manufacturer: Samsung
[128902.413801] usb 1-6.2.1: SerialNumber: 100000000d2386db
[128902.415438] uvcvideo: Found UVC 1.00 device PI4 USB Device (1d6b:0104)
```

and could use the pi as a webcam.

geerlingguy, 2020-10-22:
Thanks for posting all the feedback; I agree with @blaueente that it would be better if we could find a way to make things work with newer kernels—locking in to an old kernel is not a wonderful option for many reasons (security, other software patches, etc.).

It's so strange that I can't get it to work even on the original A+ I have though, while the Zero happily runs. Don't they use the same SoC?

geerlingguy, 2020-11-23:
Just noting the download for Raspbian older releases is over at http://downloads.raspberrypi.org/raspbian/images/raspbian-2020-02-07/

fangfufu, 2020-11-23:
You cannot use the Raspberry Pi 3 as a USB gadget. At the time of writing this is only possible on the Raspberry Pi 0, 0W, A, A+, and 4. The Pi B, B+, 2, 3, 3A+, and 3B+ lack the required pin for OTG.

https://raspberrypi.stackexchange.com/questions/71613/how-to-use-raspberry-pi-3-as-a-usb-gadget

geerlingguy, 2020-11-23:
@fangfufu - Ah, that's news to me, I thought all the Pis were able to do it. Good to know, and I think I should add a compatibility chart on the README.

starbasessd, 2020-11-23:
For the PI4's (especially after the modded usb-c power issues) look here:
https://www.raspberrypi.org/forums/viewtopic.php?t=243966

jeaton2411, 2020-12-05:
Using the old kernal work around on 4b, camera only works in Facetime on Mac, no other apps recognize the can. Works fine on PC. Interestingly using 4b alpha version of show me webcam only works on Mac if cam is opened first in Photo Booth, then extremely laggy. Scrambled vertical bars otherwise.

parski, 2020-12-14:
> Using the old kernal work around on 4b, camera only works in Facetime on Mac, no other apps recognize the can. Works fine on PC. Interestingly using 4b alpha version of show me webcam only works on Mac if cam is opened first in Photo Booth, then extremely laggy. Scrambled vertical bars otherwise.
I have the same problem.

kolewu, 2020-12-19:
Have you considered insufficient power consumption and powered the 4B over GPIO instead of USB?
USB2 is limited to 500mA and the bare bone consumption of the 4B is [documented](https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md) as 600mA.

vicary, 2020-12-24:
@kolewu The only way it could work for a Raspberry Pi 4 is a powered USB hub with output specs that supports ≥2.5A in a single port, and that's not an easy thing to find.

In Amazon you'll most likely end up with charging stations, followed by powered USB hubs that separates *charging ports* and *data ports* and the latter only gives ≤2.0A.

I am still testing multiple brands of USB hubs at the moment.

StuartIanNaylor, 2021-01-02:
The port being used Pi 4 | USB-C 'Power' port not USB 2.0

https://www.hardill.me.uk/wordpress/2019/11/02/pi4-usb-c-gadget/

So presumably its also being connected to usb 3.0, but also isn't the problem universal and happens on all except zero for some reason?

DarkStiller, 2021-01-29:
I tried to use a Pi 3A+ for the webcam and it didn't work. I do not read a lot about the 3A+ here. I tried with the current RaspianOS (2021-01-12 - lite and desktop) and it stuck every time at swap or bluetooth during booting. After I found this article I tried the version 2020-05-28-lite. There it fully boots up, but no new device pops up on my Windows PC. During shutdown the Pi, it stuck at bluetooth device. After that I tried 2020-02-07-lite. Fully start up and shutdown but no device on my PC. Your sheet at the main page said, that I have to use a USB-A to USB-A Cable. I used one of that and it fully powered up the Pi. (I had undervoltage using it directly on my PC front panel, I solved the problem with an active powered USB hub)

Is there a mistake on my side or did I missed to read something here?

fschulze, 2021-01-29:
@DarkStiller this was already answered above: https://github.com/geerlingguy/pi-webcam/issues/5#issuecomment-732456971 It isn't possible on 3A+ due to lack of necessary hardware support.

Torxed, 2021-01-30:
Not sure if this is relevant, just started tinkering with it so haven't really had time to dissect this fully.
But running `pi@raspberrypi:~/uvc-gadget$ ./uvc-gadget -f1 -s1 -r0 -u /dev/video1 -v /dev/video0` causes:
```
V4L2 device is mmal service 16.1 on bus platform:bcm2835-v4l2
V4L2: Getting current format: MJPG 1280x720
V4L2: Setting format to: MJPG 1280x720
V4L2: Getting current format: MJPG 1280x720
v4l2 open succeeded, file descriptor = 3
UVC: device open failed: No such file or directory (2).
```

And checking `pi@raspberrypi:~/uvc-gadget$ ls -l /dev/video*` there's no `video1`, not sure there ever should be.
```
crw-rw----+ 1 root video 81, 3 Jan 30 12:36 /dev/video0
crw-rw----+ 1 root video 81, 0 Jan 30 12:36 /dev/video10
crw-rw----+ 1 root video 81, 1 Jan 30 12:36 /dev/video11
crw-rw----+ 1 root video 81, 2 Jan 30 12:36 /dev/video12
```

This is on a Raspberry Pi 4 Model B from 2018, using OS Version 10 (Buster):
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

It does boot however, which is a good sign.
And I ran the ansible playbook from here *(https://github.com/geerlingguy/pi-webcam)*.
Can't say I'm a big ansible fan speaking of `I love good documentation, but I hate misleading or hard-to-read documentation.`, because it's hard to actually understand what's going on in detail here and what needs to be changed. But I guess a windows "next next next" thing is what people want.

On the cient-side (the desktop that wants to use the webcam), I get the following after the ansible playbook:
```
82177.671654] xhci_hcd 0000:0f:00.3: Timeout while waiting for setup device command
[82177.878319] usb 5-4.1: device not accepting address 15, error -62
[82177.968324] usb 5-4.1: new high-speed USB device number 16 using xhci_hcd
[82183.004993] xhci_hcd 0000:0f:00.3: Timeout while waiting for setup device command
[82188.338326] xhci_hcd 0000:0f:00.3: Timeout while waiting for setup device command
[82188.544970] usb 5-4.1: device not accepting address 16, error -62
[82188.545159] usb 5-4-port1: unable to enumerate USB device
```

All I had to do was add `modules-load=dwc2,g_serial` at the end of `/boot/cmdline.txt`.
After which on the client-side using the raspberry pi as it's webcam, turned into:
```
[82307.768359] usb 5-4.1: new high-speed USB device number 17 using xhci_hcd
[82307.877691] usb 5-4.1: New USB device found, idVendor=0525, idProduct=a4a7, bcdDevice= 4.19
[82307.877694] usb 5-4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[82307.877695] usb 5-4.1: Product: Gadget Serial v2.4
[82307.877696] usb 5-4.1: Manufacturer: Linux 4.19.57-v7l+ with fe980000.usb
[82307.951274] cdc_acm 5-4.1:2.0: ttyACM0: USB ACM device
[82307.951756] usbcore: registered new interface driver cdc_acm
[82307.951757] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```

And I'm one step closer to resolving this issue.
To fix the `/dev/ttyGS0: cannot open as standard input: No such file or directory` is simply did `sudo systemctl disable getty@ttyGS0.service` as it's not really needed, there's already a raspberry os auto-login on `getty@tty1.service` in my case:
```
pi@raspberrypi:~$ ls -l /etc/systemd/system/getty.target.wants/getty*
lrwxrwxrwx 1 root root 34 Jul 10  2019 /etc/systemd/system/getty.target.wants/getty@tty1.service -> /lib/systemd/system/getty@.service
```

I still don't know what `/dev/video1` comes from, so I'll have to dig deeper.

Torxed, 2021-01-30:
Some resources on the topic that summarized my random input better:
 * http://www.davidhunt.ie/raspberry-pi-zero-with-pi-camera-as-usb-webcam/
 * https://www.hardill.me.uk/wordpress/2019/11/02/pi4-usb-c-gadget/

And it seams like the first link has some interesting stuff to get it working on Windows.
Haven't tested it tho, other than confirmed that the above post doesn't solve it for Windows *(no clue about Mac)*.

Torxed, 2021-01-31:
Tested on Windows 10, works there as well with the following `/etc/rc.local` addition *(before `exit 0`)*.

```
!/bin/bash
mkdir /sys/kernel/config/usb_gadget/pi4
echo 0x1d6b > /sys/kernel/config/usb_gadget/pi4/idVendor
echo 0x0104 > /sys/kernel/config/usb_gadget/pi4/idProduct
echo 0x0100 > /sys/kernel/config/usb_gadget/pi4/bcdDevice
echo 0x0200 > /sys/kernel/config/usb_gadget/pi4/bcdUSB
echo 0xEF > /sys/kernel/config/usb_gadget/pi4/bDeviceClass
echo 0x02 > /sys/kernel/config/usb_gadget/pi4/bDeviceSubClass
echo 0x01 > /sys/kernel/config/usb_gadget/pi4/bDeviceProtocol
mkdir /sys/kernel/config/usb_gadget/pi4/strings/0x409
echo 100000000d2386db > /sys/kernel/config/usb_gadget/pi4/strings/0x409/serialnumber
echo "Samsung" > /sys/kernel/config/usb_gadget/pi4/strings/0x409/manufacturer
echo "PI4 USB Device" > /sys/kernel/config/usb_gadget/pi4/strings/0x409/product
mkdir /sys/kernel/config/usb_gadget/pi4/configs/c.2
mkdir /sys/kernel/config/usb_gadget/pi4/configs/c.2/strings/0x409
echo 500 > /sys/kernel/config/usb_gadget/pi4/configs/c.2/MaxPower
echo "UVC" > /sys/kernel/config/usb_gadget/pi4/configs/c.2/strings/0x409/configuration
mkdir /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0
mkdir /sys/kernel/config/usb_gadget/pi4/functions/acm.usb0
mkdir -p /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/control/header/h
ln -s /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/control/header/h /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/control/class/fs
mkdir -p /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/1080p

mkdir -p /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/1080p

cat <<EOF > /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/1080p/dwFrameInterval
5000000
EOF

cat <<EOF > /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/1080p/wWidth
1920
EOF

cat <<EOF > /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/1080p/wHeight
1080
EOF

cat <<EOF > /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/1080p/dwMinBitRate
10000000
EOF

cat <<EOF > /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/1080p/dwMaxBitRate
100000000
EOF

cat <<EOF > /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/mjpeg/m/1080p/dwMaxVideoFrameBufferSize
7372800
EOF

mkdir /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/header/h
cd /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/header/h
ln -s ../../mjpeg/m
cd ../../class/fs
ln -s ../../header/h
cd ../../class/hs
ln -s ../../header/h
cd ../../../../..
ln -s /sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0 /sys/kernel/config/usb_gadget/pi4/configs/c.2/uvc.usb0
ln -s /sys/kernel/config/usb_gadget/pi4/functions/acm.usb0 /sys/kernel/config/usb_gadget/pi4/configs/c.2/acm.usb0
udevadm settle -t 5 || :
ls /sys/class/udc > /sys/kernel/config/usb_gadget/pi4/UDC
```

It's taken from the link mentioned in the previous post.
Keeping it here for record keeping. Preferably this would be sourced from another file like mentioned in the article and not put straight into `/etc/rc.local`.

But this shows up as a `UVC Camera` as expcted on Windows 10:
[image]

lkoepsel, 2021-02-07:
I have a Mac and have not been able to get the 1080p to work. The 1080p camera shows up just as the 720p camera shows up, I'm able to login via a serial connection and the status of piwebcam looks as successful as the 720p piwebcam status, however both OBS or Photo Booth show a black screen. 

1. I followed the instructions here as to uncommenting the 1080p and commenting out the 720p etc. 
2. I looked at Torxed changes and compared to the commented code for 1080p in this repository. The only differences were using the text "PI4 USB Device" and changing the dwFrameInterval to 5000000. Which I tried this as well.

Has anyone been able to use a Pi 4/HQ Camera/1080p with a Mac?

For consideration, here are the two status screens from the successful 720p and unsuccessful 1080p:
```
## Working: 720p
pi@raspberrypi:~$ sudo systemctl status piwebcam
● piwebcam.service - Pi webcam
   Loaded: loaded (/etc/systemd/system/piwebcam.service; enabled; vendor preset:
   Active: active (running) since Sun 2021-02-07 09:38:23 PST; 52s ago
 Main PID: 366 (piwebcam)
    Tasks: 3 (limit: 4915)
   Memory: 3.3M
   CGroup: /system.slice/piwebcam.service
           ├─366 /bin/bash /home/pi/uvc-gadget/piwebcam
           ├─527 sudo /home/pi/uvc-gadget/uvc-gadget -f1 -s1 -r0 -u /dev/video1
           └─553 /home/pi/uvc-gadget/uvc-gadget -f1 -s1 -r0 -u /dev/video1 -v /d

Feb 07 09:38:23 raspberrypi systemd[1]: Started Pi webcam.
Feb 07 09:38:24 raspberrypi sudo[385]:       pi : TTY=unknown ; PWD=/home/pi/uvc
Feb 07 09:38:24 raspberrypi sudo[385]: pam_unix(sudo:session): session opened fo
Feb 07 09:38:24 raspberrypi sudo[385]: pam_unix(sudo:session): session closed fo
Feb 07 09:38:24 raspberrypi sudo[527]:       pi : TTY=unknown ; PWD=/home/pi/uvc
Feb 07 09:38:24 raspberrypi sudo[527]: pam_unix(sudo:session): session opened fo
```

```
## Not working: 1080p
pi@raspberrypi:~$ sudo systemctl status piwebcam
● piwebcam.service - Pi webcam
   Loaded: loaded (/etc/systemd/system/piwebcam.service; enabled; vendor preset:
   Active: active (running) since Sun 2021-02-07 10:30:11 PST; 11min ago
 Main PID: 345 (piwebcam)
    Tasks: 3 (limit: 4915)
   Memory: 3.7M
   CGroup: /system.slice/piwebcam.service
           ├─345 /bin/bash /home/pi/uvc-gadget/piwebcam
           ├─517 sudo /home/pi/uvc-gadget/uvc-gadget -f1 -s1 -r1 -u /dev/video1
           └─530 /home/pi/uvc-gadget/uvc-gadget -f1 -s1 -r1 -u /dev/video1 -v /d

Feb 07 10:30:11 raspberrypi systemd[1]: Started Pi webcam.
Feb 07 10:30:11 raspberrypi sudo[356]:       pi : TTY=unknown ; PWD=/home/pi/uvc
Feb 07 10:30:11 raspberrypi sudo[356]: pam_unix(sudo:session): session opened fo
Feb 07 10:30:11 raspberrypi sudo[356]: pam_unix(sudo:session): session closed fo
Feb 07 10:30:11 raspberrypi sudo[517]:       pi : TTY=unknown ; PWD=/home/pi/uvc
Feb 07 10:30:11 raspberrypi sudo[517]: pam_unix(sudo:session): session opened fo
```

Given the status is seems to be successful on both, it seems to be a Mac issue.

Thanks,
Lief

ganguin, 2021-02-15:
Did someone have any success running the gadget on the pi 4 with newer 5.x kernels? always getting the RCU error?

Side note, you can downgrade your kernel on an up-to-date raspberry OS using rpi-update. The following command sets up the latest available 4.19 kernel that works with UVC gadget:
```
rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2
```

JamesHannon, 2021-11-03:
Has anyone managed to get this going yet out of interest? I was hoping to use an RPi 0W mk2 for a webcam but naturally, that isn't supported in the known good 2020 build

StuartIanNaylor, 2021-11-06:
@geerlingguy you doing an update for the Z2 Geoff?

Torxed, 2022-01-15:
Perhaps related to recent breaking changes to the camera drivers: https://www.raspberrypi.com/news/bullseye-camera-system/
They've changed the entire way RPi talks to the HQ Camera. I can't seem to get this working on the latest version.

And they've setup a repository showcasing how to use it: https://github.com/raspberrypi/libcamera-apps

I'm to unfamiliar with RPi to get this working agin.

Torxed, 2022-01-18:
> You can either switch to using the legacy camera stack, or somehow someone/somewhere we need to get uvc-gadget support with libcamera integration.

Yea the second one I guess. First isn't really a sustainable option. I got it working with RTSP and stuff but those have a slight delay compared to uvc-gadget :)

kbingham, 2022-01-18:
If anyone is interested in working on this and has time, I'll be happy to help guide or support them on it, but I haven't had time to work on it myself yet.

Torxed, 2022-01-18:
I'd love to but would mean a lot of effort

Oceanique, 2022-02-04:
Hello kbingham, I'm really interested in a guide because I'm completely lost, I bought a RPi 4 B 2gb and everything was going well, I've plugged it into a Windows OS (also Debian) and it was recognized as a UVC camera on Windows as well on Linux. But when I tried to see the video stream, the device suddenly disappears from both OS...

I tried to install an old release of raspbian like it was told previously in this post but it doesn't boot on this RPi 4, it shows me the same screen as when there's no sd card in it.