geerlingguy/pi-webcam #7: Following the steps but does not work after plugin to pc

YunchuZhang, 2020-11-14:
Hi there,
Thanks for your writing. I want to ask when I finished all those steps and shutdown raspberry pis. When I plug it to pc, it does not show any devices in pc. Do you know what may caused this issue? 

I am using pi zero w and mac

geerlingguy, 2020-11-14:
> When I plug it to pc, it does not show any devices in pc.

So are you plugging it into a PC or a Mac? In either case, you could go into Device Manager (on Windows) or About this Mac > System Report... and look for USB devices and see if you see a Raspberry Pi device connected.

Note that it takes 30 seconds to a minute the first time you plug it into your computer before it boots and would be available.

And if you have a really slow microSD card, it could take multiple minutes before it's available. Do you have an HDMI adapter you could use to watch the Pi's output while it boots up?

YunchuZhang, 2020-11-14:
I plug it to a Mac, and no device detected by it
but when I connect to a monitor and it boots up successfully

YunchuZhang, 2020-11-14:
[image]

geerlingguy, 2020-11-14:
Can you run `sudo systemctl status piwebcam` too?

YunchuZhang, 2020-11-15:
[image]
seems working on pi? What kind of usb cable it required? Does it need to be powered with another cable?

YunchuZhang, 2020-11-15:
I change another cable and it could work!!!
But the image seems too dark, is there a position I can modified the brightness? What is the range for that I saw default number is 55

geerlingguy, 2020-11-15:
@YunchuZhang - Yay! Interesting that the cable was the problem, but glad you found that!

For the brightness, yes the default of 55 works great on macOS, and also I tested it on my Dell XPS 13 laptop and it seemed to work well in Windows 10 there. But I know some people mentioned that it should be set to `127` (or really, some value higher than the default `55`).

The way to do that would be to change the `PU_BRIGHTNESS_DEFAULT_VAL` value on this line (https://github.com/geerlingguy/uvc-gadget/blob/master/uvc-gadget.c#L91) and then recompile `uvc-gadget`. I really want that to be more easily configurable (see #4), because currently the only really easy way to do that is to fork my uvc-gadget repo and change it in your fork, then pull from there, or add a step to patch that file before it's applied on the Pi.

YunchuZhang, 2020-11-16:
what micro usb cable did you use for this? I bought two new cable from Amazon, 
https://www.amazon.com/gp/product/B077HPJJFF/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
but does not work
Any recommend?

geerlingguy, 2020-11-18:
I just used one of the old cables I have in my giant bag of USB cables I've collected over the years. I tried two different cables (one of them is Anker, the other I don't see any identifying mark), and both have worked for me.

YunchuZhang, 2020-11-18:
I found that if the cable has 'B' sign in the microusb part, it could work.
Does it mean the cable must be a usb-otg one?

YunchuZhang, 2020-11-19:
Also, when you try it with Linux Ubuntu systems, did you meet the error like this?

[image]

osilloh, 2020-11-25:
[image]

This was a great little project as I have a bunch of Pi's, cheapy camera, and the flat heat sink you see, lying around. Runs at about 50 C with camera on. The camera was about $9 on Amazon and the case about $7, the Pi around $12, so this setup was <$30 US!

I did have to check here to get it to work as I am not an engineer or coder by any stretch of my imagination.

I could not get the camera to work on a Windows desktop after trying 4 different USB to micro USB cables. So I did a separate Pi USB power supply, waited for it to power up, than plugged in a USB to micro USB cable into my PC. Bingo, that worked.

I then removed the power supply cable to see what would happen. The camera remained up! with the USB in the other(non PWR-in) port!

So I'm wondering if its a timing thing, when I check device manager, I see this:

[image]

But when I check with the single USB to MicroUSB cable in the PWR-IN port it does not appear.

Is it a timing issue? Where the camera is not yet started when the Pi is first plugged in so Windows does not detect it? Can you configure a delay to reset the code you have to bring up the camera? So Windows will see it?

Thanks you!

starbasessd, 2020-11-25:
> <img width="107" alt="UVC Camera"

No, on thePiZero, the non power is OTG and can also power the pi.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

osilloh, 2020-11-25:
I am an not an engineer, but the Pi/Camera still working with only microUSB in the non PWR-IN port is a good indication that it is accepting power. I should clarify. Does the Pi and camera need to be fully started/initialized before Windows will detect it as a camera?

starbasessd, 2020-11-25:
You're gonna love this answer:
OTG starts up very early, and for ethernet or Serial gadget, it doesn't
have to be up all the way for them to be seen by Windows. I am not sure as
to the setup for Camera Gadget, still working with it...
In fact, for a Serial Gadget, you can watch most of the boot info if you
did a connection, then told it to re-boot...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

osilloh, 2020-11-25:
ohhh. After watching the video again, I am *supposed* to use the USB port and *NOT* the PWR-IN port. 

It works perfectly on the USB port with a single cable!

YunchuZhang, 2020-11-26:
> ohhh. After watching the video again, I am _supposed_ to use the USB port and _NOT_ the PWR-IN port.
> 
> It works perfectly on the USB port with a single cable!

the cable matters for me. I use ps4 cables now and could work

YunchuZhang, 2020-11-26:
Another question is when I try with Linux, it fails and the fps shows wrong. Did you try with linux?
@geerlingguy @starbasessd 
[image]

0d0a, 2021-01-15:
my pi zero refuses to boot after updating to the latest raspberry pi os 5.4.83-v7l 1379 SMP Mon Dec !4 2020

closed 2021-05-18 by stale[bot]