geerlingguy/pi-webcam #8: Adding a mic to the webcam

pranshumalik14, 2020-11-23:
Hi,

I was wondering if it is possible to adapt the scripts you have to have an optional microphone, which would just work if you have it plugged in or have the same behavior as now if the mic's not connected. This will help make an all-in-one package for a decent performing pi webcam. I'm thinking of this [breakout board](https://www.adafruit.com/product/3421). 

Let me know, since I was thinking of making one of these. 

Thanks!

geerlingguy, 2020-11-23:
I think it would be possible, but I also know it's a little outside my own expertise. For my own use, I don't need the mic as I use an external mic in all cases (or can settle for the built-in mic on my laptop). But it would be nice to have, especially for cases where the camera might be mounted in another room or outside, and the computer inside, or if you use it with a Raspberry Pi, like the Pi 400, which has no internal mic.

pranshumalik14, 2020-11-23:
Makes sense. Is it possible to add a service that makes the audio available to the host device? This breakout board has some supporting libraries and after an install and reboot, the mic will be connected on boot

pranshumalik14, 2020-11-23:
Maybe you can give me a few pointers to how I can add such a service if it's not possible to add in the changes right now. After testing it out I can make a pull request

geerlingguy, 2020-11-23:
@pranshumalik14 - That would actually require changes to the uvc-gadget project that is used—see this file for starters: https://github.com/geerlingguy/uvc-gadget/blob/master/multi-gadget.sh

pranshumalik14, 2020-11-23:
Thanks, I'll look into it!

closed 2020-11-24 by pranshumalik14