geerlingguy/pi-webcam #9: Test on rPi B (256 MB)

txt-file, 2020-11-23:
I tried with the http://downloads.raspberrypi.org/raspbian/images/raspbian-2020-02-07/ on a rPi4 and it worked. Then I switched the SD to my rPi B (256 MB) model and it doesn't work. The systemd service is running and uvc-gadget is constantly using 100 % cpu. If it works with a rPi A (without +) it should work with a B, shouldn't it?

Otherwise great timing for doing the youtube video. I was searching for a webcam last weekend and was very disappointed that I can get a rPi cam HQ for ~50 € but no regular 4k webcam for that price.

geerlingguy, 2020-11-23:
@txt-file unfortunately, according to some posts I found today over in #5, it looks like the original model B does _not_ have the proper pins for OTG usage. Currently only the 4, A/A+, and 0/0W can work with this project.

closed 2020-11-23 by geerlingguy

txt-file, 2020-11-23:
thanks @geerlingguy … I did not yet stumble over that specific https://github.com/geerlingguy/pi-webcam/issues/5#issuecomment-732456971
Time to search for a 0 or A then. :-)

starbasessd, 2020-11-23:
> stumble over that specific #5 (comment)

Did Red Shirt Jeff try it with any of the Compute Modules?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

geerlingguy, 2020-11-24:
@starbasessd - No, he has not... it doesn't seem like OTG is supported on the current IO board design though—it would need to be done through a different interface board.