geerlingguy/pi-webcam #11: Feature Request: Pre-baked image for SDCard

bradwood, 2020-11-25:
Hey @geerlingguy,

Have you considered using this: https://github.com/mkaczanowski/packer-builder-arm to pre-bake an image with the playbook changes alread applied?

I've had some good experiences with the above, so I thought I'd thow it in there.

Nice work, btw!

bradwood, 2020-11-25:
Something like this: https://gitlab.com/b-k8s/packer/-/blob/master/boards/raspberry-pi-4/ubuntu_server_20.04_arm64.json#L50 can be used to trigger the ansible as part of the image baking process... 

For bonus points, have Gitlab Actions create releases with the images in them so we can just `curl` them down and `dd` them onto the SD card... :grinning:

geerlingguy, 2020-11-25:
I may entertain this idea, but there's already a project using buildroot to make an even snazzier version of the camera software that does that, over at https://github.com/showmewebcam/showmewebcam :)

This project is a little lighter weight in terms of 'just running on top of a standard Raspberry Pi OS install', but that project is a little more turnkey.

closed 2021-03-25 by stale[bot]