geerlingguy/pi-webcam #14: Low FPS in Google Meet

szwecja, 2020-12-01:
Hi @geerlingguy,

Great project. I have made it work on my Mac. It works perfectly with Zoom, FaceTime. I have problems with:
- Google Meet in the browser (both Chrome and Safari): the frame rate is very low (ca. 1 fps) and latency is also around 1 second.
- Skype and MS Teams seem to crash instantly when I change it to Pi Webcam.
Do you also have those issues?

Thanks,
Tomasz

geerlingguy, 2020-12-01:
I haven't had those issues; Google Meet was actually the most reliable/best experience on the Pi itself in Chromium.

Have you tried rebooting the Mac (and thus also rebooting the camera at the same time) and seeing if that helps?

szwecja, 2020-12-01:
Tried to restart all, checked on another Mac. Changing into lower resolution in Meet (to 360p) does not help. It is pretty strange, since in Zoom and FaceTime it works perfectly.

geerlingguy, 2020-12-01:
@szwecja - That is very strange! Can you log into the Pi via SSH and view the logs from the piwebcam service (`journalctl -f -u piwebcam` then Ctrl-C to stop watching the logs). Maybe something in the camera negotiation is kind of blowing up. Definitely a strange issue that I haven't seen before!

szwecja, 2020-12-01:
Dec 01 16:46:43 raspberrypi-kamera systemd[1]: Started Pi webcam.
Dec 01 16:46:44 raspberrypi-kamera sudo[274]:       pi : TTY=unknown ; PWD=/home/pi/uvc-gadget ; USER=root ; COMMAND=/home/pi/uvc-gadget/multi-gadget.sh
Dec 01 16:46:44 raspberrypi-kamera sudo[274]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 01 16:46:48 raspberrypi-kamera sudo[274]: pam_unix(sudo:session): session closed for user root
Dec 01 16:46:49 raspberrypi-kamera sudo[387]:       pi : TTY=unknown ; PWD=/home/pi/uvc-gadget ; USER=root ; COMMAND=/home/pi/uvc-gadget/uvc-gadget -f1 -s1 -r0 -u /dev/video1 -v /dev/video0
Dec 01 16:46:49 raspberrypi-kamera sudo[387]: pam_unix(sudo:session): session opened for user root by (uid=0)

szwecja, 2020-12-01:
`Dec 01 16:46:43 raspberrypi-kamera systemd[1]: Started Pi webcam.
Dec 01 16:46:44 raspberrypi-kamera sudo[274]:       pi : TTY=unknown ; PWD=/home/pi/uvc-gadget ; USER=root ; COMMAND=/home/pi/uvc-gadget/multi-gadget.sh
Dec 01 16:46:44 raspberrypi-kamera sudo[274]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 01 16:46:48 raspberrypi-kamera sudo[274]: pam_unix(sudo:session): session closed for user root
Dec 01 16:46:49 raspberrypi-kamera sudo[387]:       pi : TTY=unknown ; PWD=/home/pi/uvc-gadget ; USER=root ; COMMAND=/home/pi/uvc-gadget/uvc-gadget -f1 -s1 -r0 -u /dev/video1 -v /dev/video0
Dec 01 16:46:49 raspberrypi-kamera sudo[387]: pam_unix(sudo:session): session opened for user root by (uid=0)
^C
`

szwecja, 2020-12-01:
```
Dec 01 16:46:43 raspberrypi-kamera systemd[1]: Started Pi webcam.
Dec 01 16:46:44 raspberrypi-kamera sudo[274]:       pi : TTY=unknown ; PWD=/home/pi/uvc-gadget ; USER=root ; COMMAND=/home/pi/uvc-gadget/multi-gadget.sh
Dec 01 16:46:44 raspberrypi-kamera sudo[274]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 01 16:46:48 raspberrypi-kamera sudo[274]: pam_unix(sudo:session): session closed for user root
Dec 01 16:46:49 raspberrypi-kamera sudo[387]:       pi : TTY=unknown ; PWD=/home/pi/uvc-gadget ; USER=root ; COMMAND=/home/pi/uvc-gadget/uvc-gadget -f1 -s1 -r0 -u /dev/video1 -v /dev/video0
Dec 01 16:46:49 raspberrypi-kamera sudo[387]: pam_unix(sudo:session): session opened for user root by (uid=0)
```

szwecja, 2020-12-01:
I have also noticed, that in Meet for first second the image is good, then it is getting choppy...

szwecja, 2020-12-01:
And on Windows laptop it works OK...

geerlingguy, 2020-12-01:
@szwecja - Hmm, one other question: How do you have the camera plugged in? Two things I can think of to try:

1. If it's plugged in through a hub, try plugging in direct (or through a different hub).
2. Try a different micro USB cable. I know one other person had issues (different than yours) that were solved by using a different cable.

szwecja, 2020-12-01:
Tried all of that :) plus deinstalling AdBlocks, etc. No luck. I will do a fresh reinstall of the MacOS during the weekend (wanted to do it anyway) and will let you know. Thanks.

maparker, 2020-12-03:
Came here just to report the same thing and saw that I was not the only one with the issue. I had a couple meetings today and only saw the low FPS in the Google Meet meetings. On macOS Big Sur, have not yet tested on the Windows side.

szwecja, 2020-12-03:
Also just to let you all know, did Google Meet on completely fresh Big Sur install and video was still low DPS.

starbasessd, 2020-12-03:
Out of curiosity do you have an older Mac, and does it happen there, too?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

szwecja, 2020-12-03:
Not really. I have MacBook Pro 15' 32GB RAM from 2018 with Radeon Pro Vega 16 / 4 GB HBM2. Zoom and others works perfectly.

mackemint, 2020-12-04:
Related behavior here, on a Sierra hackintosh with dedicated GPU.
Facetime is flawless, Discord misbehaves though. Using Brave browser (Chromium-based) the FPS is really low and choppy.
Using the Discord desktop app, the camera works great initially but after about a minute or so the resolution and framerate drop. 
On my other system (2015 MBP) running Mojave I can confirm the same behavior for Brave browser (slow and choppy video) but using the camera with the Discord app has perfect performance.

Checking the logs through a serial connection (no wifi on the pi) produces the same logs as for @szwecja

silverman, 2020-12-14:
I have the same issue with instant Teams crashing on a modern Mac (2017 MacBook Pro on macOS 10.15.7). It is a way different problem from the current title of this issue. Do they have a common cause? Or does it make sense to create a separate issue?

GoToMeeting also has trouble, but it's less of an instant crash so I can't as easily duplicate it all day long. It only started in GoToMeeting after I switched the aspect ratio to 16:9.

mackemint, 2020-12-14:
I can confirm the same problem that @silverman reported. This is applicable for both 720p and 1080p Pi Webcams.
I've extracted the log from Teams but I can't make much sense of it. The only hint that I can find are these lines that show up here and there. 
deviceManager: getSelectedDevices: {"camera":"0x141000001d6b0104","microphone":"AppleHDAEngineInput:1B,0,1,0:1","speaker":"AppleHDAEngineOutput:1B,0,1,1:0"}
hidDeviceManager: [hidDeviceManager]: enableHidSelectedDeviceFixForWin1809Build is true, recompute selected mic
hidService: [registerCallbacksForSpecificDeviceEvents]: Invalid or non-hid device, unable to register callbacks

Attaching diagnostics log from Teams
[MSTeams Diagnostics Log 2020-12-14_21_46_46.txt](https://github.com/geerlingguy/pi-webcam/files/5691590/MSTeams.Diagnostics.Log.2020-12-14_21_46_46.txt)

geerlingguy, 2020-12-16:
@mackemint - Ha, someone _just_ emailed me the same note this morning, and he said that it seems to have fixed the issue in Teams for him, for over a week.

It would be great to hear from others if that solves any issues, and if so, maybe the default could be changed...

maparker, 2020-12-16:
This change resolved the problem for me.

silverman, 2020-12-17:
The 720p version of the solution of @mackemint fixes the Teams problem for me!

szwecja, 2020-12-17:
@geerlingguy Indeed the change suggested by @mackemint did change everything. Now both Teams and Google Meet work with correct video. I am running in 720p on MacBook Pro. Thanks all!!

mackemint, 2020-12-17:
Tried this with 1080p settings. All the issues I experienced previously seem resolved now

- Discord used to have Frame rate dropouts after about 2 minutes of the camera being on, no longer present
- Teams app: I can select the camera without Teams crashing with "there was a glitch"
- Teams in Brave browser: framerate no longer choppy and seems perfectly smooth
- Google Meet in Brave browser: used to have the same problem as Teams which is now resolved.
- Photo booth: Camera only produced a black image - resolved
- OBS: I can select 1080p, 30 FPS instead of 1080p, 2 FPS.

Yay!

mackemint, 2020-12-17:
To everyone watching this, if anyone of you have the opportunity to clone my fork and check if my 1080p modifications work out of the box for you too, geerlingguy can merge this into main of his repo.
My pull request:
https://github.com/geerlingguy/uvc-gadget/pull/4

Install instructions are identical, just clone my repo and follow the other install instructions as usual. :)
Swap this line: git clone https://github.com/geerlingguy/pi-webcam.git 
For this line: https://github.com/mackemint/pi-webcam.git

trek2710, 2020-12-18:
Had same issue with only Teams crashing on Mac (through USB hub). Changing dwFrameInterval to 333333 solved it.

szwecja, 2020-12-23:
Thank you all for solving this issue. Merry Christmas to all!!

geerlingguy, 2021-01-18:
I believe this has been fixed by https://github.com/geerlingguy/uvc-gadget/pull/5 — thanks!

closed 2021-01-18 by geerlingguy