geerlingguy/pi-webcam #16: camera rotation

mdinata, 2020-12-04:
First, thanks for simplifying the process and awesome tutorial on youtube. It works in Pi0W without any problems. I use Pi0 case which position the case in the up-right to get the picture in the right orientation. Physical camera rotation is not possible considering how the camera module is fitted in the case. Would u please share how to rotate the camera orientation from configuration so i can put the camera sideways (USB cable from below).

Thank you

trek2710, 2020-12-18:
I believe adding this line to the file /uvc-gadget/piwebcam just after the other v4l-ctl control options should rotate the picture 90 degrees
/usr/bin/v4l2-ctl -c rotate=90

geerlingguy, 2020-12-30:
@trek2710 - If that option works—I wonder if setting `rotate=0` would be a good default (so people searching the code for `rotate` would at least find it and be able to change it to 90, 180, etc.)?

aardvarkyvark, 2020-12-31:
The issue is that the user won't know what the rotation is until they plug it in. Are there any edits they can make to the config file?

doctorleach, 2020-12-31:
The option does work, BTW.

absynth47215, 2021-04-22:
I'd vote for including this option somehow in the installation process (either by clearly documenting it or by making it configurable in the playbook). I have a Zero W in the official case and I had exactly this issue. In an ideal world, the camera would notice its own orientation, but that doesn't work in an environment without orientation sensor.

closed 2021-08-20 by stale[bot]