geerlingguy/pi-webcam #18: Feature request: mirror camera output to HDMI port.

iampegram, 2020-12-05:
It would be nice to be able to use this as an HDMI camera as well.

mackemint, 2020-12-05:
I guess you could do this with a script that runs VLC on boot?
https://www.raspberrypi.org/forums/viewtopic.php?t=80518

iampegram, 2020-12-05:
Would this work without a desktop environment installed? I'm using a Pi Zero W with Raspberry Pi OS lite to save on overhead...

starbasessd, 2020-12-05:
Could with OMXPlayer installed...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

iampegram, 2020-12-08:
I'm trying to figure that out, how do I pipe /dev/video0 into omxplayer? Do I have to send it through raspivid first? ffmpeg?

mackemint, 2020-12-11:
The suggested solution at the end here suggests using an intermediary fifo file:
https://www.raspberrypi.org/forums/viewtopic.php?t=69322

trek2710, 2020-12-18:
Got it somewhat working. First kill the uvc-gadget, then run the following command:
#vlc --color v4l2:// :v4l2-dev=/dev/video0 :v4l2-width=640 :v4l2-height=480 :v4l-brightness=60 --no-qt-privacy-ask --fullscreen
There is quite a lag on my hdmi monitor. You can also add more debugging to vlc (-vvv)
I did also have the hdmi monitor plugged in while booting and used the power usb input on the PiZ.

closed 2021-04-18 by stale[bot]

jsilence, 2021-04-21:
It is possible to output the video to HDMI, but then it seemingly can not be used as a webcam any more. 

* Configure the pi with `raspi-config` to autologin to CLI
* Place `raspivid -t 0` as the last command into the `.bashrc` of the _pi_ user.