geerlingguy/pi-webcam #19: Suggestion: Use gcc optimizations for building uvc-gadget

ErikPitko, 2020-12-05:
I tried building uvc-gadget with this configuration in Makefile:

```Makefile
CFLAGS          := -W -Wall -O3 $(KERNEL_INCLUDE)
LDFLAGS         := -O3
```
I don't see any dropped frames. CPU utilization is also pretty low.

mackemint, 2020-12-10:
@ErikPitko I can confirm that this works and builds correctly and that performance seems fine. 
Connecting it to a Windows machine works, but I can't see any improvement in image quality.
Connecting to MacOS produces no image (black image)

I produced two SD cards, one with your optimisations and one "vanilla" Geerling build that I can swap between easily for testing purposes.

When using it as an input in OBS on MacOS, with your build optimisations I can select a 1080 resolution that produces a black image only. 
Don't have the possibility to try the same on the Windows system, can you confirm that your configuration actually produces a 1080 input or if it just defaults to a lower resolution?

ErikPitko, 2020-12-10:
I only tested in on Linux machine with the following tool: https://webcamtests.com/ which produced the following output:

[image]

I also tried the fps test which showed that its not dropping any frames.

Have you changed the uvc-gadget/piwebcam script to allow 1080p streaming as well as the uvc-gadget/multi-gadget.sh script?

mackemint, 2020-12-10:
@ErikPitko oh I missed that file actually, I only made changes to multi-gadget.sh
Thanks, I'll try that out.

mackemint, 2020-12-10:
Still no good, what did I miss?
I updated the config.yml to clone my fork here:
https://github.com/mackemint/uvc-gadget/blob/master/piwebcam

ErikPitko, 2020-12-10:
@mackemint I tried cloning and testing your fork, its working for me. I am building uvc-gadget on the rpi zero itself, maybe thats what's causing the problem?

```bash
$ git clone https://github.com/mackemint/uvc-gadget.git
$ make
$ sudo reboot -f
```

I will try to connect it to windows machine later. Unfortunately I'm not able to get to one right now.

mackemint, 2020-12-10:
Interesting! I was building on a Raspberry Pi 2, I'll do it properly and report back.

mackemint, 2020-12-11:
Finally got this working!
Cloned my pi-webcam repo on the RpiZero and followed Geerlings build instructions. This produced the same results as I described earlier, possible to select 1080 but only producing a black image output.
After cleaning the uvc-gadget repo on the Pi and building again according to your last comment, the unit showed up as Pi Webcam #2 
I was then able to select 1080 in OBS successfully.
https://webcamtests.com/ produces everything from 2FPS 480p to 1080 depending on web browser, USB cable and room brightness. 😆 

Thanks for your input ! 🙇

geerlingguy, 2021-01-18:
I believe this has been fixed by https://github.com/geerlingguy/uvc-gadget/pull/5 — thanks!

closed 2021-01-18 by geerlingguy