geerlingguy/pi-webcam #22: Not working for me on Mac

mannix, 2021-01-08:
Hi @geerlingguy, thank you for providing this repo!

I followed the instructions to set things up. I'm on a Pi Zero W connected to a MacBook Pro. It powers on and I can SSH on to it (I don't have a micro hdmi to connect it to a monitor.) When I check About this Mac > System Report, the camera is not showing up.

On the Pi, when I run `sudo systemctl status piwebcam` I get:
```
🔴 piwebcam.service - Pi webcam
   Loaded: loaded (/etc/systemd/system/piwebcam.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2021-01-08 21:55:09 GMT; 12min ago
  Process: 302 ExecStart=/home/pi/uvc-gadget/piwebcam (code=exited, status=1/FAILURE)
 Main PID: 302 (code=exited, status=1/FAILURE)

Jan 08 21:55:08 raspberrypi sudo[433]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jan 08 21:55:09 raspberrypi piwebcam[302]: V4L2 device is mmal service 16.1 on bus platform:bcm2835-v4l2
Jan 08 21:55:09 raspberrypi piwebcam[302]: V4L2: Getting current format: JPEG 1024x768
Jan 08 21:55:09 raspberrypi piwebcam[302]: V4L2: Setting format to: MJPG 1280x720
Jan 08 21:55:09 raspberrypi piwebcam[302]: V4L2: Getting current format: MJPG 1280x720
Jan 08 21:55:09 raspberrypi piwebcam[302]: v4l2 open succeeded, file descriptor = 3
Jan 08 21:55:09 raspberrypi piwebcam[302]: UVC: device open failed: No such file or directory (2).
Jan 08 21:55:09 raspberrypi sudo[433]: pam_unix(sudo:session): session closed for user root
Jan 08 21:55:09 raspberrypi systemd[1]: piwebcam.service: Main process exited, code=exited, status=1/FAILURE
Jan 08 21:55:09 raspberrypi systemd[1]: piwebcam.service: Failed with result 'exit-code'.
```

It looks like `UVC: device open failed: No such file or directory (2).` might be a clue, but I'm not sure how to debug that further. 

Any tips on what else I can look at to troubleshoot?

Thank you!

mikeprevette, 2021-01-10:
Same issue here, Pi Zero, piCamera v2. Will investigate

mikeprevette, 2021-01-11:
What happens when you run uvc-gadget/piwebcam directly? 

I get - 
`mkdir: cannot create directory ‘/sys/kernel/config/usb_gadget/pi4’: File exists
mkdir: cannot create directory ‘/sys/kernel/config/usb_gadget/pi4/strings/0x409’: File exists
mkdir: cannot create directory ‘/sys/kernel/config/usb_gadget/pi4/configs/c.2’: File exists
mkdir: cannot create directory ‘/sys/kernel/config/usb_gadget/pi4/configs/c.2/strings/0x409’: File exists
mkdir: cannot create directory ‘/sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0’: File exists
mkdir: cannot create directory ‘/sys/kernel/config/usb_gadget/pi4/functions/acm.usb0’: File exists
ln: failed to create symbolic link '/sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/control/class/fs/h': File exists
cat: write error: Device or resource busy
cat: write error: Device or resource busy
cat: write error: Device or resource busy
cat: write error: Device or resource busy
cat: write error: Device or resource busy
cat: write error: Device or resource busy
mkdir: cannot create directory ‘/sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/header/h’: File exists
ln: failed to create symbolic link './m': File exists
ln: failed to create symbolic link './h': File exists
ln: failed to create symbolic link './h': File exists
ln: failed to create symbolic link '/sys/kernel/config/usb_gadget/pi4/configs/c.2/uvc.usb0/uvc.usb0': Operation not permitted
ln: failed to create symbolic link '/sys/kernel/config/usb_gadget/pi4/configs/c.2/acm.usb0/acm.usb0': Operation not permitted
ls: write error: Device or resource busy
V4L2 device is mmal service 16.1 on bus platform:bcm2835-v4l2
V4L2: Getting current format: MJPG 1280x720
V4L2: Setting format to: MJPG 1280x720
V4L2: Getting current format: MJPG 1280x720
v4l2 open succeeded, file descriptor = 3
`

mannix, 2021-01-11:
looks the same for me: 
```
pi@raspberrypi:~/uvc-gadget $ ./piwebcam
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
mkdir: cannot create directory '/sys/kernel/config/usb_gadget/pi4': File exists
mkdir: cannot create directory '/sys/kernel/config/usb_gadget/pi4/strings/0x409': File exists
mkdir: cannot create directory '/sys/kernel/config/usb_gadget/pi4/configs/c.2': File exists
mkdir: cannot create directory '/sys/kernel/config/usb_gadget/pi4/configs/c.2/strings/0x409': File exists
mkdir: cannot create directory '/sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0': File exists
mkdir: cannot create directory '/sys/kernel/config/usb_gadget/pi4/functions/acm.usb0': File exists
ln: failed to create symbolic link '/sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/control/class/fs/h': File exists
cat: write error: Device or resource busy
cat: write error: Device or resource busy
cat: write error: Device or resource busy
cat: write error: Device or resource busy
cat: write error: Device or resource busy
cat: write error: Device or resource busy
mkdir: cannot create directory '/sys/kernel/config/usb_gadget/pi4/functions/uvc.usb0/streaming/header/h': File exists
ln: failed to create symbolic link './m': File exists
ln: failed to create symbolic link './h': File exists
ln: failed to create symbolic link './h': File exists
ln: failed to create symbolic link '/sys/kernel/config/usb_gadget/pi4/configs/c.2/uvc.usb0/uvc.usb0': Operation not permitted
ln: failed to create symbolic link '/sys/kernel/config/usb_gadget/pi4/configs/c.2/acm.usb0/acm.usb0': Operation not permitted
ls: write error: Device or resource busy
V4L2 device is mmal service 16.1 on bus platform:bcm2835-v4l2
V4L2: Getting current format: MJPG 1280x720
V4L2: Setting format to: MJPG 1280x720
V4L2: Getting current format: MJPG 1280x720
v4l2 open succeeded, file descriptor = 3
UVC: device open failed: No such file or directory (2).
```

mannix, 2021-01-11:
I'm diving in to the `piwebcam` script and the scripts it calls to see what I can learn.

mannix, 2021-01-11:
I don't know if it's a red herring or not, but "UVC: device open failed: No such file or directory (2)." it output because it's looking for `/dev/video1` which does not exist on my pi.

geerlingguy, 2021-01-11:
@mannix - Do you have the camera enabled in the `/boot/config.txt` (set either manually or via `sudo raspi-config`, or via the interface in Pi OS)?

mannix, 2021-01-11:
I thought it was. I had previously ran `sudo raspi-config` and enabled it. However, when I run `raspistill -v -o test.jpg` I get 
```
mmal: main: Failed to create camera component
mmal: Camera is not enabled in this build. Try running "sudo raspi-config" and ensure that "camera" has been enabled
```
I ran it again and see the confirmation: 
[image]

Then I rebooted (based on raspi-config's prompt). And now the `raspistill` command works. However I'm still getting the same error I shared in my first comment.

Stragely, the entire contents of `/boot/config.txt` is now
```
pi@raspberrypi:/boot $ cat /boot/config.txt
start_x=1
gpu_mem=128
```

I had checked it before the `raspi-config` run and it was full of the usual stuff. 🤔

mannix, 2021-01-12:
I started over from scratch. This time I made sure the camera was enabled before running the ansible script. I have plugged it into both my Mac and a Windows machine and neither one recognizes it. I am using the correct usb port on my Pi (I tried both just to be sure.) When I SSH onto it I do have what appears to be a healthy system:

[image]

This is unfamiliar territory for me so I'm not sure what else to check at this point.

Also, I recognize this has become more of a support thread rather than an actual issue (although perhaps it will uncover one.) I appreciate your help but if you feel I would be better served in the Raspberry Pi forums I'll be happy to do so.

mannix, 2021-01-13:
I'm continuing to try and troubleshoot this. At this point I think it could be that my MacBook is not recognizing the Pi in "gadget mode". I wonder if it's because of the usb dongle/adapter I'm using. Other USB type A devices work with this dongle but maybe for some reason the Pi won't.

Has anyone gotten this to work on a MacBook? How are you connecting it?

geerlingguy, 2021-01-13:
@mannix - I would see if you can get it working plugged in direct to your Mac (no USB hub or adapter if possible), and try one or two different cables (some cables, for whatever reason, just have issues).

Also make sure you wait at least 20-30s for the Pi to boot, and make sure you're plugging it into the proper port (not the PWR port).

mannix, 2021-01-13:
I just ordered a microUSB -> USB C cable that should arrive tomorrow. Fingers crossed that's all I need!

mikeprevette, 2021-01-13:
My issue seemed to be related to also trying to run a ethernet gadget (g_ether) at the same time as a composite config as I was on a non wifi Zero. I switched to a wifi Zero, removed g_ether and got it to work finally.

mannix, 2021-01-18:
Once my new cable came (direct micro usb to usb c - avoiding the adapter) it worked. Thanks for all the help!

closed 2021-01-18 by mannix