geerlingguy/pi-webcam #23: 'pi webcam' appears in photo booth, bot selecting it deselects all cameras

Itai-Nelken, 2021-01-09:


Itai-Nelken, 2021-01-09:
https://user-images.githubusercontent.com/70802936/104090264-59bf5380-527e-11eb-8550-6283dd810f3c.mov
here is a screen recording, when clicking on the pi webcam option the picture from the "real webcam" freezes.

Itai-Nelken, 2021-01-09:
I'm using a pi zero W with the official camera v2 (not the HQ one).

BeauSlim, 2021-01-14:
I have a v2 also.  I see the same problem in Photo Boot but the camera works fine for me in FaceTime and Duo (in Chrome).   
Maybe PhotoBooth is fussy about resolutions.  I remember seeing the same thing after plugging in an old HDV camera via FireWire.  HDV has an odd 1440x1080 resolution.

Itai-Nelken, 2021-01-14:
Thanks! I'll try to use FaceTime later.
I forgot to mention it doesn't work with cheese on a linux machine as well.

yoyellow, 2021-01-18:
Had the same issue in photobooth, but the camera does work in OBS, Quicktime or Facetime..

Im using the original raspberry Pi Cam 1.3. works like a charm and alot more affordable.. lens angle is good for webcam use.

stevewallin, 2021-03-08:
I have found that for photobooth, WebEx and some other apps they seem to have a limit on the bitrate of the stream they can manage at about 1.2mbps.... updating the config and lowering that had solved all of the compatibility issues for me.

closed 2021-09-04 by stale[bot]