geerlingguy/pi-webcam #24: uvc-gadget causes 100% cpu when camera is idle [rpi-zero-w]

dhensen, 2021-01-09:
(Just made this rpi zero HQ camera following your Youtube + this github together with my 5yo daughter. She is very happy, so thanks for that @geerlingguy for bundling this into an ansible playbook.)

Question: when I'm not using the camera (usb cable is connected to my computer, but not actively using the camera) I see 100% cpu usage mainly because of uvc-gadget. When I start using the camera the cpu usage drops to around 20%.

Any idea why this is so? I'm using a rpi Zero W if it matters.

```
  CPU[|||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||100.0%]   Tasks: 33, 7 thr; 1 running
  Mem[|||||||||||||||||||||||||||||||||||||||||||||||||||||||||||                                        39.5M/368M]   Load average: 1.15 1.16 1.02 
  Swp[                                                                                                    0K/100.0M]   Uptime: 01:34:43

  PID USER      PRI  NI  VIRT   RES   SHR S CPU% MEM%   TIME+  Command
  386 root       20   0  3660  2824  2756 R 93.7  0.7 32:23.42 /home/pi/uvc-gadget/uvc-gadget -f1 -s1 -r0 -u /dev/video1 -v /dev/video0
  580 pi         20   0  8344  3072  2264 S  2.5  0.8  0:48.88 htop
  772 pi         20   0  6332  3352  1900 R  2.5  0.9  0:01.83 htop
```

dhensen, 2021-01-09:
Oh... I should have looked at upstream first before posting here. I just found this: https://github.com/wlhe/uvc-gadget/issues/1
I'm closing because it's not an issue for this project!

closed 2021-01-09 by dhensen

geerlingguy, 2021-01-10:
@dhensen - Heh, see also: https://github.com/geerlingguy/uvc-gadget/issues/1