geerlingguy/pi-webcam #27: Fatal Error Running Ansible Playbook

grabvba, 2021-01-19:
Followed the guide, went to run the main.yml cmd and got the following error, 

fatal: [127.0.0.1]: FAILED! => {"reason": "couldn't resolve module/action 'make'. This often indica                                                       tes a misspelling, missing collection, or incorrect module path.\n\nThe error appears to be in '/ho                                                       me/pi/pi-webcam/tasks/uvc-gadget.yml': line 9, column 3, but may\nbe elsewhere in the file dependin                                                       g on the exact syntax problem.\n\nThe offending line appears to be:\n\n\n- name: Build uvc-gadget.\                                                       n  ^ here\n"}

Don't see what the issue could be so im a little stuck.

geerlingguy, 2021-01-19:
@grabvba - How did you install Ansible? If the version is too old or if you installed only `ansible-base` (and not full `ansible`) then that can happen.

grabvba, 2021-01-19:
Thanks for the reply, used the `sudo apt update && sudo apt install -y python3-dev python3-pip libyaml-dev libffi-dev git && sudo pip3 install --no-binary pyyaml ansible` command from your readme in the terminal of the pi. I've tried it a couple of times, everything seems to pull and install fine, it isn't throwing any errors but the yaml still wont run.

pavelcucka, 2021-01-20:
I have the same issue.

Houdini311, 2021-01-29:
I have the same issue as well.

sidkdbl07, 2021-03-10:
Me too. Here is my ansible install. I installed it with the same command as @grabvba above.
ansible --version
# ansible 2.10.6
#   config file = /home/pi/pi-webcam/ansible.cfg
#   configured module search path = ['home/pi/.ansible/plugins', ' /usr/share/ansible/plugins/modules']
#   ansible python module location = /usr/local/lib/python3.7/dist-packages/ansible
#   executable location = /usr/local/bin/ansible
#   python version = 3.7.3 (default, Jul 25 2020, 13:03:44) [GCC 8.3.0]

jmsiener, 2021-03-18:
I've hit a similar wall as well.  Similar issue as described above.  I'm trying a pi3 A+ if it makes any difference.  Similar output as above but ansible 2.10.7.

closed 2021-07-16 by stale[bot]