geerlingguy/pi-webcam #28: Feature Request: GPIO LED when camera is in use

pmottram, 2021-01-19:
Like most webcams, I'd like to have a LED that illuminates when the camera is recording video, and goes out once it is not.

Ideally a configurable GPIO pin - how easy is this?

geerlingguy, 2021-01-20:
That's actually a great idea, not sure why I didn't even think of it. It doesn't seem impossible, but it would probably need to be part of the uvc-webcam project that this project sets up/clones.

pmottram, 2021-01-20:
Looking a bit upstream suggests it is an option (-p) in the uvc-gadget package:

```
Usage: ./uvc-gadget [options]

Available options are
    -b value       Blink X times on startup (b/w 1 and 20 with led0 or GPIO pin if defined)
    -f device      Framebuffer device
    -h             Print this help screen and exit
    -l             Use onboard led0 for streaming status indication
    -n value       Number of Video buffers (b/w 2 and 32)
    -p value       GPIO pin number for streaming status indication
    -r value       Framerate for framebuffer (b/w 1 and 30)
    -u device      UVC Video Output device
    -v device      V4L2 Video Capture device
    -x             show fps information
```

pmottram, 2021-01-20:
A little further digging leads to 'piwebcam' file being the source of the run command for the uvc-gadget, so I think I should be able to configure the options on the line:
`sudo /home/pi/uvc-gadget/uvc-gadget -u /dev/video1 -v /dev/video0`
to
`sudo /home/pi/uvc-gadget/uvc-gadget -u /dev/video1 -v /dev/video0 -p 23`

Now to go find myself a LED to test with....

pmottram, 2021-01-20:
OK, so I repaced the uvc-gadet with the one from https://github.com/peterbay/uvc-gadget and then I could use the -p flag to enable the LED on a GPIO pin of my choice. I also added the -b flag so it blinks at me once it has booted up anbd is ready to go :)

Not sure of the differences to the version forked uvc-gadget, but it seems to run fine with my v2.1 camera module.

geerlingguy, 2021-01-20:
Let's leave this issue open, just so if someone wants to dig in a little further and add a PR to document how to do it, or modify the uvc-gadget fork (or help make sure upstreams/downstreams get synced so it works) to make it work out of the box.

pmottram, 2021-01-27:
I eventually stumbled across https://github.com/showmewebcam/showmewebcam/releases which is a super uptimised Pi webcam. Now got it up and running:

[image]

Posted the case stl here: https://www.thingiverse.com/thing:4731152

plj5003, 2021-09-21:
So I have a pimoroni blinkt (url below) sitting on top of my zero w cam (Thanks BTW! It's been rock solid for months!).  What would I look for to assess the status of the camera (i.e being on or off?).  I figure i'll just use their python lib to do all the led commands and look somewhere to see the on/off flag to set the leds like every 5 seconds or something.  Thoughts?

[https://shop.pimoroni.com/products/blinkt](url)

wytr, 2021-11-14:
Did something similar but worked my way around over the serial port, lol.
[image]

plj5003, 2021-12-14:
just following up here - is there not a way to run a command to see if it's on or not?

starbasessd, 2021-12-14:
Answer #2 shows how to test for a GPIO pin status (on/off-high/low) Is that what you are looking for?
https://raspberrypi.stackexchange.com/questions/53778/how-to-detect-state-of-gpio-pin-in-python

plj5003, 2021-12-14:
sorry to see if the webcam is streaming / recording vs idle