geerlingguy/pi-webcam #29: Webcam not working on Windows 10 with Teams or Skype

ThiemoCh, 2021-01-21:
I have a very strange Problem with Teams and Skype on Windows 10:
The Camera is detected after startup and usable via the Windows Camera App.
I can setup the Camera on both Teams and Skype correctly.
As soon a I start a call (both Teams and Skype) everything looks normal to me, I can see myself through the Pi-Webcam as well as the counter-party.
However they can not see me, they only see my profile Picture, as if I would not be using a Webcam.
I tried everything that came to my mind - even testing it on different PCs, but no luck.
I don't even know how I could debug this problem.
Maybe someone faced the same problem already or has suggestions where to dig deeper into the configuration.
Cheers,
Thiemo.

ThiemoCh, 2021-02-10:
For anyone interested, the Problem is non existing when using showmewebcam:
https://github.com/showmewebcam/showmewebcam

LevelZero-dev, 2021-05-13:
This sounds like the same issue as #14 
I applied the workaround listed there and haven't had the issue since.

closed 2021-09-10 by stale[bot]