geerlingguy/pi-webcam #44: ansible-playbook: command not found

Chris3636, 2021-05-15:
got as far as 
cp inventory-local.example inventory, OK so far
ansible-playbook main.yml
then,
bash: ansible-playbook: command not found
I`m pretty new to this so go easy on me !!
Any advise would be gratefully recieved

geerlingguy, 2021-05-15:
You will need to install Ansible before you can use that command: https://docs.ansible.com/ansible/latest/installation_guide/intro_installation.html

Chris3636, 2021-05-15:
I thought I already did that with the command
sudo apt update && sudo apt install -y python3-dev python3-pip libyaml-dev libffi-dev git && sudo pip3 install --no-binary pyyaml ansible
at an earlier stage

geerlingguy, 2021-05-15:
Can you run `which ansible`? And also `sudo pip3 freeze` and paste that output here?

Chris3636, 2021-05-15:
pi@raspberrypi:~ $ sudo pip3 freeze
ansible==2.7.7
apache-libcloud==2.4.0
asn1crypto==0.24.0
astroid==2.1.0
asttokens==1.1.13
automationhat==0.2.0
bcrypt==3.1.6
beautifulsoup4==4.7.1
blinker==1.4
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
certifi==2018.8.24
chardet==3.0.4
Click==7.0
colorama==0.3.7
colorzero==1.1
cookies==2.2.1
cryptography==2.6.1
cupshelpers==1.0
docutils==0.14
drumhat==0.1.0
entrypoints==0.3
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==1.0.2
fourletterphat==0.1.0
gpiozero==1.5.1
html5lib==1.0.1
httplib2==0.11.3
idna==2.6
isort==4.3.4
itsdangerous==0.24
jedi==0.13.2
Jinja2==2.10
jmespath==0.9.4
keyring==17.1.1
keyrings.alt==3.1.1
lazy-object-proxy==1.3.1
lockfile==0.12.2
logilab-common==1.4.2
lxml==4.3.2
MarkupSafe==1.1.0
mccabe==0.6.1
microdotphat==0.2.1
mote==0.0.4
motephat==0.0.3
mypy==0.670
mypy-extensions==0.4.1
netaddr==0.7.19
ntlm-auth==1.1.0
numpy==1.16.2
oauthlib==2.1.0
olefile==0.46
pantilthat==0.0.7
paramiko==2.4.2
parso==0.3.1
pexpect==4.6.0
pgzero==1.2
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
piglow==1.2.5
pigpio==1.78
Pillow==5.4.1
psutil==5.5.1
pyasn1==0.4.2
pycairo==1.16.2
pycrypto==2.6.1
pycups==1.9.73
pygame==1.9.4.post1
Pygments==2.3.1
PyGObject==3.30.4
pyinotify==0.9.6
PyJWT==1.7.0
pykerberos==1.1.14
pylint==2.2.2
PyNaCl==1.3.0
pyOpenSSL==19.0.0
pyserial==3.4
pysmbc==1.0.15.6
python-apt==1.8.4.3
pywinrm==0.3.0
pyxdg==0.25
PyYAML==3.13
rainbowhat==0.1.0
reportlab==3.5.13
requests==2.21.0
requests-kerberos==0.11.0
requests-ntlm==1.1.0
requests-oauthlib==1.0.0
responses==0.9.0
roman==2.0.0
RPi.GPIO==0.7.0
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
Send2Trash==1.5.0
sense-hat==2.2.0
simplejson==3.16.0
six==1.12.0
skywriter==0.0.7
sn3218==1.2.7
soupsieve==1.8
spidev==3.4
ssh-import-id==5.7
thonny==3.3.6
touchphat==0.0.1
twython==3.7.0
typed-ast==1.3.1
unicornhathd==0.0.4
urllib3==1.24.1
webencodings==0.5.1
Werkzeug==0.14.1
wrapt==1.10.11
xmltodict==0.11.0
I am getting no response to which ansible?

geerlingguy, 2021-05-16:
```
ansible==2.7.7
```

That's a really old version of Ansible! I'm guessing you may have had it installed previously.

The best solution would be to completely reinstall:

```
sudo pip3 uninstall ansible
sudo pip3 install ansible
```

Make sure to do that in two commands so it upgrades properly.

Chris3636, 2021-05-16:
pi@raspberrypi:~ $ sudo pip3 uninstall ansible
Not uninstalling ansible at /usr/lib/python3/dist-packages, outside environment /usr
Can't uninstall 'ansible'. No files were found to uninstall.
pi@raspberrypi:~ $ sudo pip3 install ansible
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: ansible in /usr/lib/python3/dist-packages (2.7.7)
pi@raspberrypi:~ $ 
Im thinking of reflashing the card and not updating the OS and running the commands again, just in case there is a conflict, what are your thoughts?

Chris3636, 2021-05-16:
UPDATE,
I reinstalled with no additional software and no updates and managed to complete the commands and assemble the camera.
The only problem I have now is that everything is completely out of focus even trying to manually adjust the lens is useless with virtually no recognisable image. For information I am running ubuntu 20.04LTS and trying to use VLC with no luck, also Cheese does not recognise the device

geerlingguy, 2021-05-16:
@Chris3636 - What kind of lens are you using? You may also need to adjust the adapter ring on the camera mount itself depending on the type of lens you're using, or even use an extension ring if you're not using one of the official lenses.

Chris3636, 2021-05-16:
I am using the official pihut one as per your video, I will experiment with some settings and try to sort it out. I am struggling to understand why cheese  does not recognise it though.

closed 2021-09-14 by stale[bot]