geerlingguy/pi-webcam #53: Mac OS Monterey does not recognise the cam

Jacubeit, 2021-11-02:
Great Project!
After updating to Mac OS Monterey, the Pi USB cam does not show up as a selectable camera anymore.
When doing a `system_profiler SPUSBDataType` it shows up as connected.
I have tried in on an intel mac and apple silicon - both on Monterey.

When attaching to a Mac on Big Sur, the same Pi/Camera shows up and works perfectly.
thx

islaeh, 2021-11-07:
FYI: two related links on the same topic (yet also without any solution): 

- The topic is also discussed in showmewebcam here https://github.com/showmewebcam/showmewebcam/issues/164
- A thread in the Apple Developer forum has also been initiated: https://developer.apple.com/forums/thread/693784

RTHPJM, 2021-11-09:
This issue is now resolved. Please re-flash your memory card with the latest version of [ShowMeWebcam (v1.91)](https://github.com/showmewebcam/showmewebcam/releases/tag/v1.91)

geerlingguy, 2021-11-09:
@RTHPJM - Thanks, just merged the PR to the uvc-gadget repo too, so it should be good to go here too. I'll let the issue stay open in case anyone finds any other issues with Monterey.

closed 2021-11-09 by Jacubeit