geerlingguy/pi-webcam #54: Webcam use on new Raspberry Pi Zero 2 W?

geerlingguy, 2021-11-03:
This issue is open to investigate whether this project works correctly on the Raspberry Pi Zero 2 W. I should've tested this earlier, but I completely forgot to do it! Has anyone else tested on one?

imp1sh, 2021-11-04:
Yes, I have one here and I installed it but it's not working.
What kind of input can I give you in order to help make this work?

RetroRock, 2021-11-06:
For me it works, but unfortunately only on Windows. On macOS nothing happens. I haven't tried Linux yet.

imp1sh, 2021-11-06:
I have revise my statement. After switching some USB cables it's now working wonderfully.
My OS is Fedora 34.

geerlingguy, 2021-11-08:
For the latest macOS, see also this bug tracked on the showmewebcam repo: https://github.com/showmewebcam/showmewebcam/issues/164

(And consider giving showmewebcam a spin, too—it's quite nice!)

JamesHannon, 2021-11-08:
I'm running OSX 12.0.1 now but I was running the previous version when I originally posted.

I tried running with the known good 2020 - Zero 2 W did not boot (assume firmware not included for that hardware)
With the latest RPi OS, my issue was no detection and a single maxed-out RPi CPU Core running the camera.

I'll burn the image again and run the setup script, try it out with my windows 10 box. see if it's a OSX thing aka https://github.com/showmewebcam/showmewebcam/issues/164 or a Pi Zero 2 W thing

htruong, 2021-11-09:
Thanks Jeff for the nice words! I'm trying to get the image to work on Pi Zero 2W, but so far it has been quite fruitless for a different reason: I can't get the image to boot.

I have put the right dtb to the right place, and uprevved the kernel, but I have been stuck with the rainbow screen.

https://github.com/showmewebcam/showmewebcam/issues/168

I have my changes pushed to the PR there as you can see, and I would appreciate a second pair of eyes to help me understand why my PR did not resulting in a booting image.

Selicos, 2021-11-09:
I see the service in a failed state so can't connect at all. New to troubleshooting Linux services but I'm pulling logs using journalctl -u piwebcam.service:

Nov 08 20:38:13 Pi2 systemd[1]: Started Pi webcam.
Nov 08 20:38:14 Pi2 sudo[362]:       pi : PWD=/home/pi/uvc-gadget ; USER=root ; COMMAND=/home/pi/uvc-gadget/multi-gadge>
Nov 08 20:38:14 Pi2 sudo[362]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Nov 08 20:38:14 Pi2 sudo[362]: pam_unix(sudo:session): session closed for user root
Nov 08 20:38:14 Pi2 piwebcam[484]: unknown control 'auto_exposure_bias'
Nov 08 20:38:14 Pi2 piwebcam[485]: unknown control 'contrast'
Nov 08 20:38:14 Pi2 piwebcam[486]: unknown control 'video_bitrate'
Nov 08 20:38:15 Pi2 sudo[487]:       pi : PWD=/home/pi/uvc-gadget ; USER=root ; COMMAND=/home/pi/uvc-gadget/uvc-gadget >
Nov 08 20:38:15 Pi2 sudo[487]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Nov 08 20:38:15 Pi2 sudo[487]: pam_unix(sudo:session): session closed for user root
Nov 08 20:38:15 Pi2 piwebcam[353]: /home/pi/uvc-gadget/piwebcam: line 13:   487 Segmentation fault      sudo /home/pi/u>
Nov 08 20:38:15 Pi2 systemd[1]: piwebcam.service: Main process exited, code=exited, status=139/n/a
Nov 08 20:38:15 Pi2 systemd[1]: piwebcam.service: Failed with result 'exit-code'.

Selicos, 2021-11-09:
Same issue, service in a failed state, on a new SD card w/ full Raspbian: 

pi@Pi2:~ $ journalctl -u piwebcam.service
-- Journal begins at Sat 2021-10-30 04:36:50 PDT, ends at Mon 2021-11-08 23:26:25 PST. --
Nov 08 23:23:48 Pi2 systemd[1]: /etc/systemd/system/piwebcam.service:6: Standard output type syslog is obsolete, automa>
Nov 08 23:23:48 Pi2 systemd[1]: /etc/systemd/system/piwebcam.service:7: Standard output type syslog is obsolete, automa>
Nov 08 23:23:49 Pi2 systemd[1]: /etc/systemd/system/piwebcam.service:6: Standard output type syslog is obsolete, automa>
Nov 08 23:23:49 Pi2 systemd[1]: /etc/systemd/system/piwebcam.service:7: Standard output type syslog is obsolete, automa>
-- Boot f206d713af0d42a7a84b6949a9c457ca --
Nov 08 23:24:27 Pi2 systemd[1]: Started Pi webcam.
Nov 08 23:24:27 Pi2 sudo[369]:       pi : PWD=/home/pi/uvc-gadget ; USER=root ; COMMAND=/home/pi/uvc-gadget/multi-gadge>
Nov 08 23:24:27 Pi2 sudo[369]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Nov 08 23:24:28 Pi2 sudo[369]: pam_unix(sudo:session): session closed for user root
Nov 08 23:24:28 Pi2 piwebcam[513]: unknown control 'auto_exposure_bias'
Nov 08 23:24:28 Pi2 piwebcam[519]: unknown control 'contrast'
Nov 08 23:24:29 Pi2 piwebcam[520]: unknown control 'video_bitrate'
Nov 08 23:24:29 Pi2 sudo[521]:       pi : PWD=/home/pi/uvc-gadget ; USER=root ; COMMAND=/home/pi/uvc-gadget/uvc-gadget >
Nov 08 23:24:29 Pi2 sudo[521]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Nov 08 23:24:29 Pi2 sudo[521]: pam_unix(sudo:session): session closed for user root
Nov 08 23:24:29 Pi2 piwebcam[354]: /home/pi/uvc-gadget/piwebcam: line 13:   521 Segmentation fault      sudo /home/pi/u>
Nov 08 23:24:29 Pi2 systemd[1]: piwebcam.service: Main process exited, code=exited, status=139/n/a
Nov 08 23:24:29 Pi2 systemd[1]: piwebcam.service: Failed with result 'exit-code'.

htruong, 2021-11-09:
The segfault seems to be a better lead. You (or someone else) needs to diagnose the crash with a debugger (ala `gdb`).

The hard part is to catch the crash at the right time. Seems like you could ssh to it, so I think compiling uvc-gadget with `-g` and then `gdb` and attach it to the process should theoretically work. But I've got my hand quite full and can't realistically assist with debugging the crash.

Selicos, 2021-11-09:
I may be able to figure out how to do that this evening or over the weekend,

htruong, 2021-11-09:
By the way, before you do that, it might be worthwhile to check out Petr Vavrin's repo to see if Petr has fixed it already:

https://github.com/peterbay/uvc-gadget

Meno-101, 2021-11-13:
Just adding to the issue, I noticed from this issue that 64-bit doesn't yet support raspistill, in which it may be the problem?

https://github.com/raspberrypi/firmware/issues/1584

gregbiv, 2021-11-15:
Same here, doesn't work on bullseye anymore :( Tried install buster instead, but had some issues installing updates

Meno-101, 2021-11-15:
@gregbiv it worked for me like a charm with this 32-bit.

The only issue is with the 1080p, I wanted to test the limits but no luck with either 64-bit nor the 1080p.

https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2020-12-04/

dieser-niko, 2021-11-20:
Can report the same. Since the camera drivers changed, it's probably the best to stick with buster, I've tried the [lite version from 28.05.2021](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2021-05-28/), works as expected

gregbiv, 2021-11-20:
Unfortunately, Debian Buster doesn't work on Pi Zero 2 W and piwebcam doesn't work on Bullseye

dieser-niko, 2021-11-20:
Can’t be true, it works on mine 

Von: Gregory Kornienko
Gesendet: Samstag, 20. November 2021 21:53
An: geerlingguy/pi-webcam
Cc: dieser-niko; Comment
Betreff: Re: [geerlingguy/pi-webcam] Webcam use on new Raspberry Pi Zero 2 W?(Issue #54)

Unfortunately, Debian Buster doesn't work on Pi Zero 2 W and piwebcam doesn't work on Bullseye

A12BA5E, 2021-11-22:
Got it work with the mentioned Buster lite version directly on the Pi Zero 2 W. For future projects there is a new kid on the block [https://www.raspberrypi.com/documentation/accessories/camera.html](url)

Meno-101, 2021-11-22:
Thank you for sharing @A12BA5E but looks like a lot of work, I would try to make it work in the weekend, if no one would got it working before then.

RavingSombra, 2021-12-08:
I just finished replicating this project on a Pi Zero 2 W with a cheap camera bought from AliExpress.
Working fine, using the Buster Lite 32bit image. For this case, had to enable legacy camera support to have video output.
Only issue is the camera needing to be rotated 90º, but that's for another topic/issue

midnightfreddie, 2021-12-20:
I just got it working on two Raspberry Pis Zero 2 W . The only issues I ran into were an apparent cable misconnect on one, and on the other I was running the playbook on a different account after already having customized the pi myself. I had to reseat the cable for the former and edit a bunch of files for the latter. I could have also just installed from the pi user for the latter, but I'm stubborn.

Tested on Win10 and MacOS Big Sur through a powered USB hub.

RonaldSerne, 2022-01-03:
Latest Pi Lite OS on Pi Zero 2 W playbook fault.
pi@WebCam01:~/pi-webcam $ ansible-playbook main.yml
Traceback (most recent call last):
  File "/usr/local/bin/ansible-playbook", line 63, in <module>
    import ansible.constants as C
  File "/usr/local/lib/python3.7/dist-packages/ansible/constants.py", line 15, in <module>
    from ansible.config.manager import ConfigManager, ensure_type, get_ini_config_value
  File "/usr/local/lib/python3.7/dist-packages/ansible/config/manager.py", line 17, in <module>
    from yaml import load as yaml_load
ModuleNotFoundError: No module named 'yaml'
pi@WebCam01:~/pi-webcam $

hrafnkelle, 2022-01-14:
Can confirm it works on RPiZW2 by using the legacy RPi OS Lite image and doing the pip3 install --upgrade pip (as root) before running the ansible part. 
Without pip upgrade ansible fails pretty quickly complaining about Yaml. Yaml will not get installed with out upgrading pip3 for some reason.

etrigan63, 2022-01-18:
Was able to install using @midnightfreddie's suggestion. I will add that you can install headlessly on the Zero 2 W by adding a configured `wpa_supplicant.conf` file along with `ssh` to the boot partition and then connecting over WiFi.

Cobaltlantern, 2022-01-20:
I can also confirm that the legacy Lite build can get this up and working on a Zero 2 W, however it did fail when I attempted to edit it to display with 1080p video in this link here:

imadoggy, 2022-02-24:
Can confirm it is working, at least for 720p, using the older version of linked here in the comments.  I split up the app installs and  did "sudo pip3 install --upgrade pip" before the last "sudo pip3 install --no-binary pyyaml ansible". My setup is pi zero 2 w, with pi HQ cam and 6mm lens. All working in win 10, haven't tested on my win 11 pc yet.

NoahTheCat, 2022-03-14:
Not working for me I'm afraid.
Pi Zero 2, HQ camera, M1 MacBook Air running 12.1 Monterey.
I followed the main install instructions, but running "sudo pip3 install --upgrade pip" as the first thing once I had SSH into the pi. everything seemed to install with no errors, but my Mac doesn't see it as a webcam.
Could this be a similar issue as faced by ShowMeWebcam? 
(https://github.com/geerlingguy/pi-webcam/issues/54#issuecomment-963119085)

imadoggy, 2022-03-14:
> Not working for me I'm afraid. Pi Zero 2, HQ camera, M1 MacBook Air running 12.1 Monterey. I followed the main install instructions, but running "sudo pip3 install --upgrade pip" as the first thing once I had SSH into the pi. everything seemed to install with no errors, but my Mac doesn't see it as a webcam. Could this be a similar issue as faced by ShowMeWebcam? ([#54 (comment)](https://github.com/geerlingguy/pi-webcam/issues/54#issuecomment-963119085))

For me I had to run "sudo apt update && sudo apt install -y python3-dev python3-pip libyaml-dev libffi-dev git" 
Then "sudo pip3 install --upgrade pip"
Followed by "sudo pip3 install --no-binary pyyaml ansible"

Essentially removing everything after && in the original instructions. Then follow the rest like normal. 
Personally I am working on tweaking the current v4l2 control settings for 720p for best quality and white balance, since 1080p still locks up the pi0 2 when trying to connect it to a windows pc. I haven't tried leaving both 720 and 1080p options, yet.

NoahTheCat, 2022-03-16:
That still doesn't work for me.  
All the individual commands seem to complete properly but when I reboot the Pi after running the Ansible playbook I can't see it as a new webcam.

Out of interest what is the difference between your commands and what I did?
It looks just like the order of the pip upgrade was changed, but why should that matter?

[image]

imadoggy, 2022-03-16:
> That still doesn't work for me. All the individual commands seem to complete properly but when I reboot the Pi after running the Ansible playbook I can't see it as a new webcam.
>
> Out of interest [...]

Looks like the way you installed it should work as well. Has any of the other UVC gadget projects worked on M1 chipset Macs yet? It sounds like the uvc driver isn't working correctly on your mac. I know this current setup and how ansible "installs" everything works for windows 10/11 and most linux OSes, but not sure about M1 macs. I would test on another computer to see if the pi webcam is functioning at all. If it works on another computer I would dig around and check for UVC gadget issues with m1 macs.

DerMarten, 2022-03-21:
The Raspberry Pi Zero 2 W with the Raspberry Pi OS Lite (**_Legacy_**) works fine for me.

imp1sh, 2022-10-11:
I cannot confirm @DerMarten. If I do that, my display manager crashes after trying to access the cam in OBS.
In VLC it sometimes just gives me one image and then freezes. Something is really weird.

nxy7, 2022-10-13:
I've just testes pi zero 2 w with 3 different rasbian versions (32-bit, 64bit and legacy) and not a single one was working, all got 'device descriptor request failed' error on win11.

Mopele, 2023-11-22:
I can not get it to run on the pi0w2 and i installed the legacy with GUI. For me I had no problem installing pyyaml and running the setup. I also see the Webcam in the device manager on windows but when I try to access the camera via Opencv or the Camera App on windows I get a error and from the Pi i can see, that the following Error on the Pi:

sudo /home/pi/uvc-gadget/uvc-gadget -u /dev/video2 -v /dev/video
  V4L2 device is unicam on bus platform: 3f801000.csi
  V4L2: Getting current format: YUYV 1280x720 V4L2: Setting format to: YUYV 1280x720
  V4L2: Getting current format: YUYV 1280x720 v412 open succeeded, file descriptor = 3
  uvc device is 3f980000.usb on bus gadget.0 uvc open succeeded, file descriptor = 4
  V4L2: Buffer mapped at address 0x76c75000, Length 1843200. V4L2: Buffer 1 mapped at address 0x76ab3000, Length 1843200.
  V4L2: Buffer 2 mapped at address 0x768f1000, length 1843200.
  V4L2: 3 buffers allocated.
  control request (req 86 cs 04) control request (req 81 cs 02)
  control request (req 86 cs 09)
  control request (req 81 cs 02) control request (req 86 cs 02)
  control request (req 82 cs 02)
  control request (req 83 cs 02) control request (req 84 cs 2)
  control request (req 87 cs 02)
  streaming request (req 81 cs 01) streaming request (req 01 cs 01) setting probe control, length = 34
  streaming request (req 81 cs 01)
  streaming request (req 83 cs 01) streaming request (req 82 cs 01)
  streaming request (req 01 cs 01)
  setting probe control, length = 34 streaming request (req 81 cs 01)
  streaming request (req 01 cs 02) setting commit control, length
  UVC: 2 buffers allocated.
  34
  V4L2: Unable to start streaming: Invalid argument (22).

diegoakel, 2024-10-27:
Firstly i had the problem with the pip install:

```

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

But then I did a virtual environment (with python3 -m venv venv) and used it to install the packages. (using pip not pip3) and its ok. But then it gives errors after at the ansible script because of the user name. I then did it manually copying each command that ansible does, but it doesnt work.

pixelpicnic, 2024-12-06:
@geerlingguy 
Is there an updated workflow for this installation?
Really appreciate the work you do!
cheers

yash-ch, 2025-06-12:
@geerlingguy 
Any Updates on this?