geerlingguy/pi-webcam #58: Could not find a version that satisfies the requirement ansible-core

drio, 2021-12-03:
Hi there, 

I am following the [setup on the pi](https://github.com/geerlingguy/pi-webcam#setup-on-the-raspberry-pi) steps on fresh raspbian [lite](https://downloads.raspberrypi.org/raspbian_lite_latest) installation and I get the following error when trying to install ansible for python (see below).

I have to reinstalling my piwebcam because osx monterry fails to [detect the cam](https://github.com/geerlingguy/pi-webcam/issues/53). 

Have you run into this issue? 

```sh
pi@piwebcam:~ $ python3 -m pip install ansible
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ansible
  Downloading https://www.piwheels.org/simple/ansible/ansible-5.0.0-py3-none-any.whl (63.3MB)
    100% |████████████████████████████████| 63.3MB 7.7kB/s

Collecting ansible-core<2.13,>=2.12.0 (from ansible)
  Could not find a version that satisfies the requirement ansible-core<2.13,>=2.12.0 (from ansible) (from versions: 0.0.1a1, 2.11.0b1, 2.11.0b2, 2.11.0b3, 2.11.0b4, 2.11.0rc1, 2.11.0rc2, 2.11
.0, 2.11.1rc1, 2.11.1, 2.11.2rc1, 2.11.2, 2.11.3rc1, 2.11.3, 2.11.4rc1, 2.11.4, 2.11.5rc1, 2.11.5, 2.11.6rc1, 2.11.6, 2.11.7rc1)
No matching distribution found for ansible-core<2.13,>=2.12.0 (from ansible)
```

geerlingguy, 2021-12-04:
Maybe try to install `python3.9` (with `sudo apt install -y python3.9` then try again? Honestly I am still on Ansible 4.x, which I believe doesn't have this new Python version requirement :(

kylergib, 2021-12-04:
you can tell pip to install ansible4 with 

sudo apt install -y python3-dev python3-pip libyaml-dev libffi-dev git && sudo pip3 install --no-binary pyyaml ansible=4.0.0

I did that and it worked fine.

drio, 2021-12-04:
Thank you both. These works:

`sudo pip3 install --no-binary pyyaml ansible==4.0.0`

☝🏽 Notice the `==` instead of `=`.

Unfortunately, OSX still does not detect the camera. If anyone did anything else to make it work in Monterrey, please let me know. I'll keep digging to see if I figure it out.

closed 2021-12-04 by drio