geerlingguy/pi-webcam #60: not working on M1 mac

ah3243, 2022-01-27:
Great project that i used before many times. However, for some reason it's only recognised for a few seconds on my new m1 mac. It will appear as a video source in zoom for example for a few seconds catch a frame but then disappear. If i use it as a source for openv (python api) then one frame is captured but all subsequent frames fail, with this error message: 
"
OpenCV: out device of bound (0-0): 1
OpenCV: camera failed to properly initialize!
"

I've tried two different cameras and am using a raspberry pi zero w v1. Do you know what could be going wrong?
thanks

JavierLuna, 2022-03-07:
I suspect the issue is with Monterey and not M1: https://developer.apple.com/forums/thread/693784
I am having this issue with two laptops: M1 pro and intel. Intel didn't had the issue until I updated to monterey

luchfilip, 2022-06-19:
still a problem, not working on M1

closed 2022-10-22 by stale[bot]

kenknoll, 2022-12-22:
Same here - not working on an M1 pro. It's showing up in the system report as USB device, yet not as camera. Any other clues? In the linked thread https://developer.apple.com/forums/thread/693784 they mention that the software needs to be changed, so this is beyond my skills, sadly. Appreciate any help!