geerlingguy/pi-webcam #61: Pi zero 1.3 cannot get device description

Sev7eNup, 2022-02-11:
Installation done. After plugging the cam into a notebooks usb interface you can hear the usb connection sound but windows says that its not possible to get the device description.

[image]

[image]

zopha, 2022-03-14:
I'm running in to this issue as well, not sure what might be happening as I don't have a screen to connect on boot. I was using the Bullseye image from the Raspberry Pi imager if that makes a difference. Going to try Legacy (Buster) tonight to see if it makes a difference.

zopha, 2022-03-19:
Legacy (Buster) worked for me but ran in to the issue described in #63 , once I was passed that it worked.. except it freezes when I try to use it. I need to try the fix menioned in #41 but moving to Buster helped with this particular issue.

jons63, 2022-04-11:
Had the same issue as @Sev7eNup and @zopha. I did not have to use fix #41 when using [legacy Buster version 2021-12-02](https://downloads.raspberrypi.org/raspios_oldstable_armhf/images/raspios_oldstable_armhf-2021-12-02/). I did run in to #63 which is an easy fix. So far everything works 👍

closed 2022-09-19 by stale[bot]