geerlingguy/pi-webcam #72: Usb device descriptor read error 110 on pi zero w

daneskildsen, 2022-07-31:
Hi there! I have followed the description for this project but I am experiencing a strange problem. I am using a Pi zero w and it is connected to my laptop which is running ubuntu. This is the output of sudo dmesg -w (from the host)

[20734.381420] usb 1-2: new full-speed USB device number 12 using xhci_hcd
[20749.909000] usb 1-2: device descriptor read/64, error -110
[20765.524782] usb 1-2: device descriptor read/64, error -110
[20765.764747] usb 1-2: new full-speed USB device number 13 using xhci_hcd
[20781.388423] usb 1-2: device descriptor read/64, error -110
[20797.008059] usb 1-2: device descriptor read/64, error -110
[20797.120080] usb usb1-port2: attempt power cycle
[20797.780028] usb 1-2: new full-speed USB device number 14 using xhci_hcd
[20803.020035] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[20808.399879] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[20808.611859] usb 1-2: device not accepting address 14, error -62
[20808.739876] usb 1-2: new full-speed USB device number 15 using xhci_hcd
[20813.771821] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[20819.151710] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[20819.359637] usb 1-2: device not accepting address 15, error -62
[20819.359803] usb usb1-port2: unable to enumerate USB device

I don't know how to debug this. Any suggestions?

imp1sh, 2022-08-12:
I'm also experiencing this problem.

ArbenP, 2022-10-09:
I too am facing this problem, on Windows 10, saying:  Unknown USB Device (Device Descriptor Request Failed)
```
Windows has stopped this device because it has reported problems. (Code 43)
A request for the USB device descriptor failed.
```

ArbenP, 2022-10-09:
Update: I have enabled Legacy Camera in raspi-config, and now it shows up on my pc, but as soon as I open the camera in any app, it works for one frame, then the pi "unplugs" from the pc and no longer shows up unable to connect again until I manually unplug then replug the raspberry pi zero w. Here's a screen recording of that in action. https://streamable.com/91b86x

daneskildsen, 2022-10-09:
@ArbenP Hi there Arben! Thanks for working on this. I still have this problem and I have not been able to solve it. So I hope that you find a solution and post it here because I would love to use the camera in OBS / Windows. Regards, Dan

daneskildsen, 2022-10-09:
Hey @geerlingguy 
Have you seen that we are still struggling with this problem, any input would be appreciated. 
Regards,
Dan

ArbenP, 2022-10-22:
Tried Raspberry Pi OS Legacy (Debian Buster) instead of the latest (Bullseye), no difference, only works for one frame then disconnects

Deovi, 2023-02-23:
I'm having the same issue

kevleyski, 2023-04-20:
Hi - just flagging I'm seeing similar too and keen to resolve
I'm powering the Raspberry Pi4 + PoE HAT over Ethernet to ensure have enough current - the device seems to be working fine against MacOS hosts (in fact the MacBookPro puts out enough current that doesn't need PoE in my case)

But when I enable UVC with Windows 10 on a regular Lenovo desktop + standard Windows install I'm getting that Code 43 (request for USB device descriptor failed) issue too

WillYouth, 2023-05-23:
same issues here on RPi zero W and raspberry pi camera module v3. Any help is greatly appreciated!!!

heLiiOs-eNg, 2023-09-15:
Are there any solutions? I still have this issue.

debloper, 2023-10-08:
I just faced this issue, and was able to solve it. Was hard to find the root cause, but here it is...

The  `uvc-gadget/piwebcam` script (invoked by the piwebcam service) tries to set the source and sink video devices, but fails, because the device IDs that it has hardcoded (i.e. `/dev/video0` as source and `/dev/video1` as the sink) may not necessarily be the case with the system you're trying to set it up.

For my case, the sink was needed to be set to `/dev/video2` instead.

You can verify if this is the case by checking `service piwebcam status` and reading its log (or, by manually checking the output of `sudo /home/pi/uvc-gadget/uvc-gadget -f1 -s1 -r1 -u /dev/video1 -v /dev/video0`).

To know which device to set output to, in that case, run:
```bash
# list all the video devices
v4l2-ctl --list-devices
# will show something like this:
# 3f980000.usb (gadget.0):
#        /dev/video2
#
# unicam (platform:3f801000.csi):
#         /dev/video0
#         /dev/video1
#         /dev/media3

# check the capabilities of the devices
v4l2-ctl -d /dev/video2 --all
# make sure the Device Caps show Video Output as listed capability, like this:
#         Device Caps      : 0x04200002
#                Video Output    <----- Make sure this is not "Metadata Capture" (if so, check next device)
#                Streaming
#                Extended Pix Format
```

Once you've found the device that can work as the sink, update that device to the script (instead of `/dev/video1`), save the script and reboot.

After restarting, the service should start without hiccup (do verify that) and the USB should enumerate as well.

Hope this helps.