geerlingguy/pi-webcam #82: error : externally-managed-environment

Oceanique, 2024-02-07:
Hello,

I succesfully, in 2022, set up a Raspberry Pi HQ camera with a RPi 4B, it was working beautifully (I was planning to use it for astronomy).

After two years without connecting it to my PC which I changed the OS from Ubuntu to Linux Mint, the camera wasn't detected with lsusb.

I suspected the OS to be the problem so I switched back to the latest Ubuntu, but still nothing. So I formatted the SD card from the RPi4B, installed the last "lite 64 bits" release and I have this message showing up : 

**error : externally-managed-environment 

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian packaged Python package,
create a virtual environment using pythton3 -m venv path/to/venv.
Then use path/to/new/bin/python and path/to/new/bin/pip. Make
sure you have python3-full installed.

For more information visit http://rptl.io/venv

note : If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.**

What should I do ? Thanks.

starbasessd, 2024-02-07:
As far as RPiOS, you may need to go back to Buster or Bullseye, as Bullseye and Bookworm have major changes over Buster and camera software and Python. It is well documented on the RPi Foundation forums. I believe a lot of the changes went upstream to Debian and over to all the other Debian based OSs...

Oceanique, 2024-02-07:
Thank you, I'm really a newbie in RPi world, I will do that thanks ! I'll get this post updated :)

Oceanique, 2024-02-07:
Hm just to be clear, you meant going back to Buster on my main computer or to Buster on the RPi ? Thank you

starbasessd, 2024-02-07:
Buster on the Pi.

Oceanique, 2024-02-07:
Thank you, I was on Deepl to clarify the "as far as RPiOS", my English isn't very good ^^.

Oceanique, 2024-02-08:
With Bullseye, there's no error showing up (only telling me "permission denied" when entering the last command "ansible-playbook main.yml", I use sudo and it works fine.).

But on Ubuntu Ubuntu 22.04.3 LTS, with dmesg -w command, it returns this : 

[50888.674864] usb 2-2: new high-speed USB device number 22 using xhci_hcd
[50893.814931] usb 2-2: device descriptor read/64, error -110
[50909.430943] usb 2-2: device descriptor read/64, error -110
[50909.539097] usb usb2-port2: attempt power cycle
[50909.954846] usb 2-2: new high-speed USB device number 23 using xhci_hcd
[50915.190979] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[50920.566886] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[50920.774827] usb 2-2: device not accepting address 23, error -62
[50920.902906] usb 2-2: new high-speed USB device number 24 using xhci_hcd
[50925.942993] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[50931.318976] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[50931.526927] usb 2-2: device not accepting address 24, error -62
[50931.527049] usb usb2-port2: unable to enumerate USB device

And it's not detected with lsusb...

starbasessd, 2024-02-08:
Did you try different USB ports, or a self-powered USB hub? Do you have other USB devices plugged in?

Oceanique, 2024-02-08:
> Did you try different USB ports, or a self-powered USB hub? Do you have other USB devices plugged in?

Yes I tried the three of them and there's no other USB devices plugged in, I still get the same errors with dmesg -w with Ubuntu 20.04.6 LTS

Oceanique, 2024-02-09:
Wait a minute... Is it normal to see the desktop or the RPiOS lite waiting for me to log in on my monitor when I restart it ? It's like it's doing nothing, the "pi-webcam" line is showing up though. I kinda remember it wasn't the case two years ago...

closed 2024-08-15 by github-actions[bot]